SmokeMonsterPacks/Super-NT-Jailbreak #71: Glitches: Front Mission (USA) shaking text on maps

SmokeMonsterPacks, 2018-03-13:
I'm copying this bug report over from [Nerdly Pleasures blog](https://nerdlypleasures.blogspot.com.br/2018/02/super-nt-review-ideal-fpga-console.html). Could someone please verify this and post their ROM's hash?

> Front Mission (shaking text box on map screens sometimes)

## firmware version and SD card model
- Jailbreak firmware version: Occurs on all firmware revisions

SmokeMonsterPacks, 2018-03-13:
RaspberryAlpine couldn't reproduce this bug via SD2SNES

ROM Tested: Front Mission - Gun Hazard (J) [T+Eng1.01_AGTP].smc 
SNT Firmware: 4.4

SmokeMonsterPacks, 2018-03-18:
Kev has a fix for Front Mission coming in the next firmware update.

SmokeMonsterPacks, 2018-03-27:
Closed until someone tests in JB v6.6. Please comment if it needs to be re-opened.

SmokeMonsterPacks, 2018-03-29:
Great Hierophant is still getting shaking text when loading Front Mission via SD2SNES. It does not occur when the ROM is via SD in the Jailbreak FW however.

Could someone please test the real cart to see if the bug is SD2SNES-specific?

G-lithcz, 2018-04-20:
The bug can be easily reproduced by starting up Front Mission (not Front Mission: Gun Hazard - totally separate game) from a cartridge and waiting 15 seconds or so before a text box appears on the world map. The text will be shaking. If it is not shaking restart the game and it should appear after one or two restarts.

SmokeMonsterPacks, 2021-02-24:
Please test in the latest fw

tjanas, 2021-02-25:
I saw the shaking text with the latest firmware.

GunbusterZero, 2021-02-27:
I am on latest JB 7.0 firmware, can confirm shaking text too.

tjanas, 2022-01-10:
Apparently this cartridge wasn't an officially licensed cartridge, but rather a very old unofficial translation on a repro cart.

closed 2022-01-10 by tjanas