SmokeMonsterPacks/Super-NT-Jailbreak #72: Glitches: Flashback - The Quest for Identity (Various regions) Audio buzz in inventory

SmokeMonsterPacks, 2018-03-13:
I'm copying this bug report over from [Nerdly Pleasures blog](https://nerdlypleasures.blogspot.com.br/2018/02/super-nt-review-ideal-fpga-console.html). Could someone please verify this and post their ROM hash?

> Flashback - The Quest for Identity (audio buzz heard when opening inventory, affects all official versions except the English/French PAL version, which behaves with no buzz.)
Regions affected: USA: En,Fr,De; Japan; Europe: En,Fr,De
Regions unaffected: Europe: En,Fre

- Jailbreak firmware version: Occurs on all firmware revisions

scotthammack, 2018-03-13:
Confirmed that the USA and Europe versions work as described. The buzz is present in the USA and Europe (En,Fr,De) versions but silent in Europe (En,Fre). I did not have a chance to test the Japanese version.

Hashes:
064a880a8dfcf576f74ae8a17c3ec7b0a27e8cb0300a5e5959452fcc30422f14  Flashback - The Quest for Identity (USA) (En,Fr,De).sfc
a47d73a5494d009cb8ea825093699ce9105b93f3f538d7bac86544024af093e  Flashback (Europe) (En,Fr).sfc
6cce3d0c8b2eec350475e3c274d7ad80c9a208ba101d9cf9ac637c5d09760ab7  Flashback (Europe) (En,Fr,De).sfc

JB 6.5

SmokeMonsterPacks, 2018-03-13:
Thanks, Scott!

SmokeMonsterPacks, 2018-03-19:
Kev has a fix for Flashback coming in the next official firmware update.

SmokeMonsterPacks, 2018-03-26:
This has been fixed in v4.4

closed 2018-03-26 by SmokeMonsterPacks