SmokeMonsterPacks/Super-NT-Jailbreak #73: Glitches: Marvelous - Mouhitotsu no Takara-jima (JP) text glitches

SmokeMonsterPacks, 2018-03-13:
I'm copying this bug report over from [Nerdly Pleasures blog](https://nerdlypleasures.blogspot.com.br/2018/02/super-nt-review-ideal-fpga-console.html). Could someone please verify this and post their ROM hash?

> Marvelous - Mouhitotsu no Takara-jima (text discoloration, SA-1 chip game)

- Jailbreak firmware version: Occurs on all firmware revisions

Confirmed by Glitchz on [Discord](https://discord.gg/EX57xnF), who provided pictures:
[image]
Super NT

[image]
Real Super Nintendo (2-1-3)

gnadenlos, 2018-03-13:
I can confirm the problem. 

As it's a SA-1 game there are no working roms. Only Japanese carts or reproductions with English translation. My reproduction is from retrocircuits - it's a SA-1 4in1 cart.

frederic-mahe, 2018-03-13:
If I am not mistaken, My Life in Gaming's Super Nt review reported that bug for the Japanese version of that game (see https://youtu.be/d_OW_t9RXEM?t=7m57s). Analogue and kevtris are aware of the problem, so let's hope it will be fixed in the official firmware.

SmokeMonsterPacks, 2018-03-18:
Kev has Marvelous fixed in the next official firmware update.

SmokeMonsterPacks, 2018-03-26:
This has been fixed in v4.4

G-lithcz, 2018-04-22:
Taking a closer look at this issue it appears it may be unresolved. Before the FW version 4.5 fix, there was anti-aliasing on every single letter [single letter ](https://user-images.githubusercontent.com/33355340/37319045-36441d84-264c-11e8-847a-23b88335b263.jpg)

treyturner, 2018-04-23:
I noticed this too on a RetroCircuits SA-1 4-in-1 multicart. It's better than it was before 4.5, but still not 100% fixed. Thanks for categorizing and detailing it so well.

James-F2, 2018-08-05:
@G-lithcz you are wrong about the 2/1/3 SNES.
Your 'converter' clips the light grey shades into white, it's the crap one from ebay (I have it too).
The regular SNES 3Chip or 1Chip renders the 512x224 text **EXACTLY** like Higan does, or should I say the other way around.

Here is a 1Chip SNES through the OSSC in 5x screenshots.
* Notice the grey lines around white lines (same thickness), which is exactly how Higan renders them.

[image]
[image]
[image]
[image]

GreatHierophant, 2018-08-05:
Here is further evidence of James-F2 and my claim :

http://www.mediafire.com/file/2n8w44dfw8cin91/Marvelous_Captures.zip/file
http://www.mediafire.com/file/a1l1gn8p39rur48/Marvelous_Photos.zip/file

The first folder has my captures of Marvelous running on a Super Nt, through original hardware and through Higan v1.06.  Note how the gray and white areas of the text align almost precisely between my S-Video capture and Higan.  

The second folder has all of James-F2's pictures he took of his screen using his 1chip and the OSSC in4x and 5x mode.  For reference I have included photos of Super Nt and Higan on my monitor in the root.

In short, our argument is that the Super Nt is not doing enough anti-aliasing on this text.

fisso, 2019-07-31:
Fixed by Paulb-nl. Do Color math on every half cycle with current and previous color. 
[HiColor Math Fix](url)

MP2E, 2019-07-31:
All contributions to MiSTer are submitted under the GPLv3 which makes it dubious legal ground at best to use the MiSTer as reference since none of the SNT firmwares are open source

closed 2021-02-24 by SmokeMonsterPacks