SmokeMonsterPacks/Super-NT-Jailbreak #74: Glitches: Top Gear 3000 (PAL, Australia) - (Minor issue) Incorrect colour @ 1 player split screen & after race completion (Track Description screen, random)

tmagri, 2018-03-13:
## Expected behaviour and actual behaviour
Within the race, instead of the colour for the outlier of font and border colour, being black, goes green. (For the first race this occurs when computer Player 2 arrives @ 3/4 on Lap 1 (21 to 22 seconds in roughly)). 
[image]

After race completion (sometimes, not frequently), background colour instead of being black the colour goes blue.
[image]

## Steps to reproduce the problem
Once the game is loaded:
	1. CHAMPIONSHIP
	2. 1 PLAYER SPLIT
	3. RACE
	4. Press A to exit TRACK DESCRIPTION (Issue 2, Blue, doesn't happen here)
	5. Go down to EXIT
	6. Race as per usual or stay idle. Either way, when computer Player 2 arrives @ 3/4 on Lap 1 (21 to 22 seconds in roughly), the black outlier of font, and border changes from black to green.
	7. Once the race is complete, and continue pressing A until TRACK DESCRIPTION is shown, it may (or may not) have a blue background instead of black (continue more races until issue occurs). 

## ROM name and hash value

Unable to test ROM dump due to not having facilities to dump the cartridge.
I am unsure of the similarity between what roms that currently available like Top Gear 3000 (EUR).sfc* compared to Australia cartridge (although cartridge ID contains EUR) 

Since Cartridge contains DSP-4 Enhancement, the Jailbreak cannot run the ROM.

From testing Top Gear 3000 (EUR).sfc* on Higan (RetroArch Accuracy), with using dsp4.rom, issues do not occur:

*MD5: 46‌2e‌15‌c0‌18‌c6‌53‌33‌5b‌a6‌34‌d3‌c1‌cd‌06‌c9
SHA1: 777D8732023A300A2C566F2D48BC354E0AA1DA7E

Untested on SD2SNES due to not having the device, this would be a good comparison to observe.

## firmware version and SD card model

- Jailbreak firmware version(s):  all, including latest snt_firmware_verJB6.5, and the official 4.4
- relevant Super Nt settings
	• Hardware: PAL (50Hz)
	• Resolution set at 1080p50Hz (the issue also occurs at 60Hz, outside of scope)
	• Any shader configuration (My preference: No Shader)
	• Any Screen size (My preference: 8x7)
	• Any Buffer mode (My preference: Zero Delay)
- Cartridge ID: SNSP-A3TP-EUR
- Phone Shots / Cartridge Scans
[image]
[image]
[image]
[image]
[image]

frederic-mahe, 2018-03-13:
@tmagri excellent bug report, thanks! If you can, could you please check whether or not the bug occurs on a genuine Super Nintendo?

SmokeMonsterPacks, 2018-03-13:
There is only one PAL region ROM, so that is good for others testing via SD2SNES. I just tested TG3K on my Retro Duo Portable via SD2SNES and didn't see it. RDP is not a perfect proxy of real hardware, but I would agree that this looks like a SNT bug.

SmokeMonsterPacks, 2018-03-13:
RaspberryAlpine on Discord just verified both glitches in the EU ROM via SD2SNES

Settings: 1080p60, PAL

tmagri, 2018-03-14:
@frederic-mahe Hi, Just to check (although confirmed), the issues are not present on the SNES PAL console. 
[image]
[image]
[image]
[image]
[image]
[image]

SmokeMonsterPacks, 2018-03-19:
Kev has a fix for Top Gear 3000, but it won't be in the next official firmware update. He wants to release it in a future release to isolate if its fix causes any new issues.

Chronokeys, 2019-08-16:
Hey, guess the fix hasn't been implemented yet, just wanted to confirm that this still happens on the latest JB with a real NTSC cart, both the blue background screen and the green border during races.

teknomedic, 2020-08-04:
Similar bug happens using FXPak Pro with Space being blue when on planet/track selection screen.  Also after several races the text reporting your winnings is garbled (Think in area 3 when it first appeared) .  Perhaps this is a special chip problem.

teknomedic, 2020-09-01:
Update:  For those that care.  The green border glitch and the blue space background glitches are still happening on the FXPak Pro.  Having re-tested on the Super NT (latest jailbreak) with no cart and using the latest firmware on last revision of the SD2SNES (regular) this no longer happens; however, using the SD2SNES Pro / FXPak Pro the split screen green border and blue screen glitches happen again.  This is using a NTSC/USA ROM from No intro's 2020 updated packages.  If I can ever remember where I put my Original SNES I'll test the SD2SNES and FXPak Pro's on that and update.

tjanas, 2021-05-29:
@tmagri does this bug still exist with JB v7.0?

teknomedic, 2021-05-29:
I'll have to test it again... I haven't played it on the Super NT since
getting a MiSTer.  Though I've also seen the same issue on MiSTer.

tjanas, 2023-05-04:
@teknomedic @Chronokeys can you please re-test your cartridges with the latest official firmware? https://www.analogue.co/support/super-nt/firmware/5.1

tmagri, 2023-05-06:
@tjanas : I had booted up and updated the firmware of the Super NT, with the Top Gear 3000 game.
Issue still persists.
[image]
[image]

tjanas, 2023-05-09:
This is an outstanding issue with [MiSTerFPGA](https://github.com/MiSTer-devel/SNES_MiSTer/issues/290) too.

tjanas, 2023-05-09:
Per kevtris regarding this game issue: "they do an insane amount of DMAs and it can get DMA-bound. when that happens it can then corrupt the palette"

BarbieOnWeed, 2023-05-10:
Wasn't all known issues fixed in the new firmware?
It say so on the site.

tjanas, 2023-07-11:
@BarbieOnWeed more issues are fixed in [JB v7.2](https://github.com/analoguejb/Analogue-Super-Nt-JB), especially those that are JB-specific.

BarbieOnWeed, 2023-07-11:
I have tested the game from SD on JB v7.2 and it seems fixed! :)

tjanas, 2023-07-12:
I have also found that the green border issue is absent when running from SD card but present when running from an sd2snes.

Chronokeys, 2023-07-12:
Issue still present when running the real ntsc cart, got the blue screen after the first race…bummer…