SmokeMonsterPacks/Super-NT-Jailbreak #75: Glitches: Wizardry VI - Bane of the Cosmic Forge (J)[T+Eng 99% V0.99a Tico] does not boot

treyturner, 2018-03-15:
## Expected behavior and actual behavior
Expected: Game boots, I hack, slash, and cast awesome wizard spells

Actual: Game does not boot, I sob softly in a corner 

## Steps to reproduce the problem
Boot the game via SD

## ROM name and hash value
`Wizardry VI - Bane of the Cosmic Forge (J)[T+Eng 99% V0.99a Tico].smc	
eb865c530fb7e045ac6f26463b370548491548d0
79beab3732f99794a547984b6e986943
9c0c7c6b`

I have no cartridge or other hardware to test with. The untranslated ROM does boot, so maybe it's a translation issue. The original ROM is:
`Wizardry VI - Kindan no Mahitsu (Japan).sfc
af177665def09103330de2daddd86b4a2bfbe68a
f5ae0c2811ac905f798c70a972184f43
c17ecba5`

## firmware version
v6.5

## relevant Super Nt settings
1080p/60

SmokeMonsterPacks, 2018-03-15:
I tested this ROM on my Retro Duo Portable via the SD2SNES and it boots. That my mean it boots on real hardware but not the SNT. Could someone please confirm?

frederic-mahe, 2018-03-17:
This ROM boots on a Super Famicom 1-CHIP + SD2SNES and seems to be playable.

gnadenlos, 2018-03-18:
Works fine from SD2SNES on firmware 4.4.

Rahzadan, 2018-03-24:
> Works fine from SD2SNES on firmware 4.4.

I can also confirm that this is a jailbreak-only bug. Boots and plays fine on my SD2SNES vis Super Nt fw 4.4.

SmokeMonsterPacks, 2018-03-27:
Closed until someone tests in JB v6.6. Please comment if it needs to be re-opened.

treyturner, 2018-03-31:
Same issues applies in v6.6.

Jayde6, 2019-08-23:
Same Issue applies in v6.9.

SmokeMonsterPacks, 2021-02-24:
I'm closing this because it's an unofficial romhack.

closed 2021-02-24 by SmokeMonsterPacks