SmokeMonsterPacks/Super-NT-Jailbreak #76: Glitches: Chrono Trigger crashing when saving on overworld randomly

R9delta, 2018-03-16:
**Expected behavior and actual behavior**

When attempted to save on the overworld (in this instance the first time visiting 2300 A.D.) instead of transitioning to the status screen by pressing X the game crashing. I am able to enter the Super NT's menu or soft reset it using the Super NT hotkeys but I am not able to continue the game as it is.

**Steps to reproduce the problem**

This seems to happen in certain parts of the map and only on my second save slot. As such below I have included a video I have taken of the issue replicated twice over along with an upload of my sram/sav.

Video of issue:
https://www.youtube.com/watch?v=lGfP0newJ3M

Save file:
http://www.mediafire.com/file/1v9770o47yurq7y/Chrono+Trigger+%28USA%29.sav

To clarify this issue only seems to be happening on the second save slot. I first noticed it once I hit 2300 A.D. and it seems to happen it certain spots of the map. P

lease see the video for reference.

This save file originated from my cartridge which i have dumped using a Retro Freak. It was then put on my Super NT for the first time starting with firmware 6.5 JB.

**ROM name and hash value**

Chrono Trigger (USA).sfc
MD5 a2bc447961e52fd2227baed164f729dc
SHA-1 de5822f4f2f7a55acb8926d4c0eaa63d5d989312

This also happens with my original non-repro cartridge which should match these values as the rom has no header.

**Firmware**

I have tested on both Jailbreak 6.5 and the normal 4.4 firmware

SmokeMonsterPacks, 2018-03-16:
This has been confirmed by users in a couple different places (reddit, twitch). Thanks for writing it up!

frederic-mahe, 2018-03-16:
@R9delta thanks for this detailed bug report. The ROM is valid:
```
Chrono Trigger (SNS-ACTE-USA, revision:SNS-ACTE-0)
    SHA256: 06d1c2b06b716052c5596aaa0c2e5632a027fee1a9a28439e509f813c30829a9
    SHA1: de5822f4f2f7a55acb8926d4c0eaa63d5d989312
    MD5: a2bc447961e52fd2227baed164f729dc
    CRC32: 2d206bf7
```

deanzel, 2018-03-18:
I've had the same issue playing on an original NTSC cart. Had it happen twice (not in the overworld), when switching to the menu. The first time was even just 10-15 minutes into the game while still at the fair. I started a new game save, but kept 2 of my more complete old ones on the cart as well.

SmokeMonsterPacks, 2018-03-18:
Kev has Chrono Trigger fixed in the next official firmware update.

SmokeMonsterPacks, 2018-03-26:
This has been fixed in v4.4

closed 2018-03-26 by SmokeMonsterPacks