SmokeMonsterPacks/Super-NT-Jailbreak #77: Glitches: Batman Begins

cacophony1, 2018-03-16:
Initially Boos wrote in the Discord chat: "Batman Returns off SD card and the sound effects were cutting in and out. Music played fine but punches and impact sounds would come and go."

I attempted to reproduce tonight using the Batman Begins rom (from recent Smokemonster pack), running on JB6.5, and hit various glitches:

I completed level 1-1 with no sound issues, but as soon as level 1-2 started there was no sound at all (no music or sound effects, it was completely silent). On Kevtris' suggestion I tried switching to fully buffered mode (instead of zero lag). So I did that, and on a couple attempts at 1-1 and 1-2 I didn't notice any issues. So then I switched back to zero delay and loaded the rom again, but this time it froze before I was able to start  the game. So I did a reset and then it seemed fine. I played through the 1-1 with no issues, and 1-2 seemed fine as well. But I lost my last life at the boss, and it asked me if I wanted to continue. I said yes and it immediately froze at a black screen.

At this point I tried switching back to fully buffered and after restarting the game I hit the freeze on start issue. I restarted again and it was fine and played a bit with no issues.

cacophony1, 2018-03-16:
On Discord, SegaSnatcher wrote: 
"Yikes, JB doesn't like Batman Returns.  Got a black screen first load.   Second load it booted, but Part 2 of Level 1 had no sound and it froze right at the boss."

frederic-mahe, 2018-03-16:
This is a valid ROM:
```
Batman Returns (SNS-BJ-USA, revision: SNS-BJ-0)
    SHA256: f8d5c51f74df33edc827fbf8df7aab70160770ab0a896db6e59438ad9208cc6e
    SHA1: 087f414d51c999e25ced561fe21503c559d8050f
    MD5: 5ef8e1f3e88c86f149838c569a0c0328
    CRC32: e87dfdf6
```

SegaSnatcher, 2018-03-16:
I can confirm after about 12 attempts I was able to get the actual cart of Batman Returns to freeze on Official Firmware.    It froze to black right after the Scene 1 intro title.

scotthammack, 2018-03-16:
For what it's worth, I just played through the first few levels with no issues during that time, but then I ran out of lives and had to use a continue, and after that, I have no audio. I was at the boss fight with the Strongman (level 2-3) when this happened.

JB6.5 firmware, loading from the SD card, same ROM as above.

SmokeMonsterPacks, 2018-03-19:
Kev has a fix for Batman Returns coming in the next official firmware update.

SmokeMonsterPacks, 2018-03-26:
This has been fixed in v4.4

closed 2018-03-26 by SmokeMonsterPacks