SmokeMonsterPacks/Super-NT-Jailbreak #78: hq scalers changing color pallet

masterninjah, 2018-03-16:
the hq2-4x scaler settings slightly change the color pallet.  the scale 2-3x leaves the colors the same as no scalers on.  

this happens on both real carts and roms

jailbreak firmware 6.5

leodsgaming, 2018-03-16:
This is a known issue, it's related to the fact this filters require a change in the color thing, that I don't know how it's called, and it's unfixable, or at least a really hard fix, because of the nature of the scaler.

But it's a official firmware issue, and KEvtris himself is fully aware of the issue and exactly how it works.

SmokeMonsterPacks, 2018-03-16:
I'll close this then since Kev is aware of it. Thanks @masterninjah & @leodsgaming

closed 2018-03-16 by SmokeMonsterPacks