SmokeMonsterPacks/Super-NT-Jailbreak #84: Ability to "really" run PAL games at 60Hz

ShinKaz, 2018-03-23:
as discussed in the last 5 comments here : 
https://github.com/SmokeMonsterPacks/Super-NT-Jailbreak/issues/55

and by Borti here : 
https://shmups.system11.org/viewtopic.php?f=6&t=61037&sid=9971373056c0298b2707d88e83c37d79&start=480

when playing a PAL game with the Super NT set to "any resolution"60 (eg 1080p60) the game still runs at 50Hz, but with added frames to match the 60Hz refresh rate of the monitor.

there's a mod for the actual PAL snes to force games to run at 60Hz, by setting, according to @tltx, PPU1 pin 24 and PPU2 pin 30 hi or low. ( http://www.mmmonkey.co.uk/snes-5060hz-switch-with-lockout-switch/ )

some of the games are acting up when forced to 60Hz, like Star Fox or Yoshi's Island, but it doesn't represent a big amount of games...so I guess that setting "for PAL to run at 60Hz" could be hidden when advanced mode is on ? I don't know.

case in point though, I'm only playing NTSC roms instead of my PAL carts right now because of this, which is kind of a bummer tbh.

iancourt, 2018-03-24:
Yes! This was discussed at great length on Discord. I think kevtris is aware how important PAL region with a 60Hz core is to PAL users. :)

While some PAL games do indeed go completely m balls at 60Hz, these games are generally ones which are correctly optimised. There are, however, a significant number of PAL titles which are completely unoptimised and play almost 17% slower at 50Hz than the NTSC version. Forcing the PAL version to run at 60Hz often resolves this issue.

It was suggested that there standard options of PAL/NTSC be increased to PAL/NTSC/PAL60/NTSC50 if "advanced options" is enabled.

Yes, we know NTSC50 wouldn't really be required, but it would be there for completeness I guess, since you could pretty much guarantee that someone would ask about it!

Hopefully it will be implemented in the official firmware. If it is, it will surely appear in the JB firmware too.

ShinKaz, 2018-03-25:
Looking forward to it :)

TheRealFlamepanther, 2018-03-26:
Just curious, but how would this differ from setting the hardware mode to NTSC? Region lockout?

iancourt, 2018-03-26:
@TheRealFlamepanther Yes, region lockout. It can be bypassed on some games by setting the region to PAL, running the game, then going back to the menu while the game is running, setting the region to NTSC, then exiting the menu. The PAL game will then be running on a 60Hz core.

While this is fine (albeit a little cumbersome) for games that only perform a region check at the start, there are numerous games that also check during play.

ShinKaz, 2018-03-27:
@TheRealFlamepanther , @iancourt no, like discussed in the shmup link by Borti, this il actually still running the game in PAL50, but with added frames to match the 60Hz refresh rate, not really running at native 60Hz like NTSC games do.

SmokeMonsterPacks, 2018-03-28:
Kev wants to smooth out region issues in a future official fw, so I'm going to close these to get the focus back on bugs.

closed 2018-03-28 by SmokeMonsterPacks