SmokeMonsterPacks/Super-NT-Jailbreak #87: Improvement:   Ability to save user inputted cheat codes and load them via SD Card.

SegaSnatcher, 2018-03-27:
This should fall under Official Firmware request. 

Inputting 1 or 2 line of cheat codes everytime is not a problem, but if you want to use a good amount of cheat codes I think the ability to save/load them for future use would be a very welcomed feature.

ShinKaz, 2018-03-27:
yes, please, seriously...I know there are cheat database per games, I'd love to be able to just store those in sd:/cheats/ and be done with it.

http://www.mediafire.com/file/mii0yhnkbzy/Snes9xGX+Cheats.zip
https://www.zophar.net/cheats/snes.html
http://www.cheatzilla.com/cgi-bin/czm/cheatSearch:/HTMLForm/R05VIFR5cGVkU3RyZWFtIDEApAkPuAjgpAgLo4DhpAgLowhpQ2hlYXRMaXN0ICAARFq6HsdgIA==?search=Search+by+title&platform=17&x=52&y=9

SmokeMonsterPacks, 2018-03-28:
Closing minor feature requests to shift the focus back on bugs. This issue will still be publicly visible however.

closed 2018-03-28 by SmokeMonsterPacks