SmokeMonsterPacks/Super-NT-Jailbreak #89: No SAV files when using COPYSNES

RayCommend, 2018-03-27:
## Expected behavior and actual behavior
Backup SAV files from carts.

## Steps to reproduce the problem
Insert USA carts 
Run COPYSNES with default settings.
CopySNES will dump SMC & SRM in my /COPYSNES directory. There is no SAV file in my /SAVES /SNES directory.
When starting game from SD card, there are no save files.

## ROM name and hash value
USA CARTS for:

Chrono Trigger
Final Fantasy 3
Secret of Mana
Legend of Zelda

If applicable, check against byuu's [SNES dump database](https://preservation.byuu.org/) and indicate if the bug occurs when using an original cartridge as well.

- firmware version: 
Jailbreak 6.6
## relevant Super Nt settings

SmokeMonsterPacks, 2018-03-27:
Thanks, Ray. Could someone please test these carts and see if the problem is reproducible?

SmokeMonsterPacks, 2018-03-27:
I have confirmation from Discord that Zelda and Secret of Mana dump fine, so I think there may be a problem with either your process, or carts.

closed 2018-03-27 by SmokeMonsterPacks

SmokeMonsterPacks, 2018-03-27:
Here are instructions from Crouton Zircon:
> It dumps the saves as .SRM files in the COPYSNES directory, but if you want to load them and use them with the jailbreak, they need to be .SAV files in the SAVES\SNES directory

So it's just a matter of renaming the .SRM to .SAV and moving it to where your saves directory is on the SD card.

RayCommend, 2018-03-27:
Got it. I will try that!

From: SmokeMonsterPacks
Sent: Tuesday, March 27, 2018 1:11 PM
To: SmokeMonsterPacks/Super-NT-Jailbreak
Cc: RayCommend; Author
Subject: Re: [SmokeMonsterPacks/Super-NT-Jailbreak] Glitches: CopySNES doesn'tcreate SAV for certain games (#89)

Here are instructions from Crouton Zircon:
It dumps the saves as .SRM files in the COPYSNES directory, but if you want to load them and use them with the jailbreak, they need to be .SAV files in the SAVES\SNES directory
So it's just a matter of renaming the .SRM to .SAV and moving it to where your saves directory is on the SD card.