SmokeMonsterPacks/Super-NT-Jailbreak #90: Panic Bomer World (J) - "You can not use a mouse with this game" error unless NTT Data Pad Compatibility enabled.

iancourt, 2018-03-27:
## Expected behavior and actual behavior

Game should run without error unless a mouse or other similar device is connected to the console.

## Steps to reproduce the problem

Connect a standard SNES/SFC controller. Run the game, error will be displayed.

Enable "Cartridge Direct" in "Startup Options". Power off and disconnect controllers. Power on with cart inserted, error will be displayed.

Enable "NTT Data Pad Compatibility", game will run as expected.

## ROM name and hash value

Running from cart, since it's a Japan-only SA-1 game.

Name:	PANIC BOMBER WORLD
MD5:	CC9C8195B55B94E075C771CA1628CBC8
SHA-1:	3A8793C4E03F457D7CFC3C0ADE6C67EDF091F74B

## firmware

JB6.6

SmokeMonsterPacks, 2018-03-27:
This was brought up on Discord and has been confirmed by a second user: BellyB

BellyBel, 2018-03-27:
I only have this issue when only one controller is plugged in. Using a multitap or another controller in port two also fixes it.

SmokeMonsterPacks, 2018-03-27:
@iancourt tested a box full of controllers and had the issue with all of them. It also occurs for him with no controller plugged in.

Rahzadan, 2018-04-06:
Just received my copy of this game in the mail today. I can confirm BellyBel's finidings above - the reported mouse message only appears if only 1 standard controller (or in my case, 8bitdo retro receiver) is connected. If I connect a 2nd original controller, multitap, or 8bitdo reciever to port 2, the game loads normally.

frederic-mahe, 2019-08-09:
The newly published jailbreak firmware should fix that issue. Could you please confirm @iancourt ? Thanks.

iancourt, 2019-08-10:
Confirmed, it seems to be fixed, @frederic-mahe. Thank you! :)

closed 2019-08-14 by frederic-mahe