SmokeMonsterPacks/Super-NT-Jailbreak #92: Chrono Trigger sound effect glitch when Lavos changes to final form

perilsensitive, 2018-03-28:
## Expected behavior
YouTube video of the battle with Lavos (sound effect occurs around 5:15 and lasts until 5:23)
https://www.youtube.com/watch?v=9sRU8yrcEu8

## Actual behavior
WAV file of the broken sound effect (sorry for the low volume; the easiest way for me to record this was on my phone)
https://drive.google.com/open?id=1-GFDmZpD3iNigftTUjkVLdMwKYBLKqIy

## Steps to reproduce the problem
Fight Lavos (not sure if it matters which way you get to him, as there are several).  As he is changing to his final form (the one with two small, round-ish parts on the sides and one humanoid-looking part in the middle), there is a sound effect that plays for several seconds.  The first few seconds sound OK, but after that it it starts sounding weird.  Everything else sounds OK once the effect ends, and the game otherwise plays fine.

## ROM name and hash value
Chrono Trigger (USA).sfc, sha1=DE5822F4F2F7A55ACB8926D4C0EAA63D5D989312
Firmware Version: JB6.6

I have the original cart as well, but I have not had a chance to test it yet.  I will do so later tonight and update this issue with the results.

kylophone, 2018-03-29:
Here's a save file (not mine, from r9delta on discord chat) that should get you close:

[](http://www.mediafire.com/file/1v9770o47yurq7y/Chrono+Trigger+%28USA%29.sav)

bsstephan, 2018-03-29:
Incidentally I was capturing Chrono Trigger a short while ago (2018-03-21, don't recall what version of the firmware that was, but it wasn't jailbroken) and the glitch is present. Here's a clip: https://www.youtube.com/watch?v=wiK-GlmBsW4

SmokeMonsterPacks, 2018-03-29:
Thanks, so this is not a newly introduced bug.

perilsensitive, 2018-03-29:
Confirmed that this happens on the US cart as well.

bsstephan, 2018-04-01:
We've been playing with this a bunch on the Discord, and we think we've determined that multiple SNES 2/1/3 recordings are consistent for that revision of hardware, as a reference point. They sound more like the intro "warble" and the following "screech" blend better; both are present, but it's not an immediate, hard cut from the warble to the screech.

Also, for whatever it's worth, I just tried this again on my SNt on fully buffered mode and still got the reported behavior.

SmiteOfHand, 2018-04-01:
Here is what we've found so far. -Glitchz and Rahzadan's videos are confirmed 2/1/3 SNES hardware. The last video is from a speedrunner that sounds identical so likely is also 2/1/3 but unconfirmed.

-Glitchz: https://youtu.be/tr_PNeQU7Ck
Rahzadan: https://www.youtube.com/watch?v=a1LmTYUwp4w
puwexil: https://www.twitch.tv/videos/182768412?t=05h29m45s

Here is another speedrun video (need to bump volume) that sounds like the original sample posted above. Unclear what hardware, linked just for reference.

EssentiaFour: https://www.twitch.tv/videos/6035064?t=04h46m30s

GreatHierophant, 2018-04-06:
Here is my recording of the Super Nt and then my SHVC-CPU-01 2/1/2 :

http://www.mediafire.com/file/ejqg8blrodvbqrs/CT_Lavos_Super_Nt_-_SHVC-CPU-01.flac

My original hardware recording is almost total warble, there are only a few instances of screech and they are shorter than anything posted above except for the original video.

SmokeMonsterPacks, 2018-04-06:
Thanks for all of the samples and clips. Should this issue be closed then?

SmiteOfHand, 2018-04-06:
Can't listen to flac on my phone unfortunately so will have to check that latest sample later. While I agree there does appear to be variance between original hardware, it is still a pretty wide gulf between those, or even higan, and what snt produces.

Maybe the SPC player when released will make this easier to sample/debug?

bsstephan, 2018-04-06:
I agree with the comments that this at least warrants a look into the issue despite the variance in other consoles' behavior, because the other consoles' behavior is in a relatively tight window of variance compared to the SNt. If we were talking about, like, the SNt warbled/blended but it was a second less than real hardware, I'd say yeah, close it, but the SNt is not close, and the reason it is not close may be more widespread with other examples we just haven't found yet.

Said another way, the starkness of the difference might mean a more general behavior could be improved, or at least warrants an explanation to inform whether or not it's worth fixing. IMO of course.

GreatHierophant, 2018-04-18:
User RaspberryAlpine on discord did a recording of a 1chip SNESjr playing the Chrono Trigger sound effect and it sounded more or less like other original hardware systems.  I made an SPC of the sound effect which you can play in sd2snes or any other SPC player.

G-lithcz, 2018-04-20:
This is what you should hear from real hardware -  https://instaud.io/24j2

This is what you hear from the snt - https://instaud.io/24jd

RaspberryAlpine, 2018-04-24:
SNES Jr recoding of GreatHierophant's SPC file 
https://cdn.discordapp.com/attachments/418896436428537859/435498545521360896/SNESjr_Chrono_Trigger_Lavos_2nd_to_3rd_form.mp4

Seiik, 2019-08-01:
Super NT with JB Ver6.6 - Chrono Trigger US copy.

I just had this issue right now playing on my original cart. Wanted to check an old save of mine and ended up fighting Lavos. I immediately noticed that the scream before its final form felt odd and not the unnatural to the original thing. That's how I ended up here in fact.

I'm not sure it does that specific scream only at that part of the game though. After that, I dumped my cart and save file via COPYSNES, which I can provide if needed, just let me know!

Seiik, 2021-03-08:
Just wanted to add that the issue unfortunately seems to persist on JB Ver7.0. 

Hopefully one day we will get the accurate Lavos morphing cry! :P

SmiteOfHand, 2023-05-04:
Apparently this has finally been fixed. Don't believe I have the save file still kicking around unfortunately. Anyone able to confirm?

"Fixed Chrono Trigger final boss sound."
https://www.analogue.co/support/super-nt/firmware/5.1

tjanas, 2023-05-05:
@perilsensitive @Seiik can you confirm the issue is fixed with your cartridges with the latest firmware update?

bsstephan, 2023-05-05:
I also have it on my todo list to check (I have a save near the end on my SD2SNES) once I'm not busy with work/new kid stuff. But that YouTube clip from @SmiteOfHand sounds right to me.

tjanas, 2023-05-08:
Thank you, @SmiteOfHand. I will close this issue, but it can be reopened if the original poster's results are different.

closed 2023-05-08 by tjanas