SmokeMonsterPacks/Super-NT-Jailbreak #93: Glitches: Unholy Night (U) CopySNES dump error

SmokeMonsterPacks, 2018-03-28:
Unholy Night plays from cart and SD perfectly. However, RevQuixo dumped his English Unholy Night cart via CopySNES and the resulting ROM crashes after finishing the first match. The original cart works fine though.

MD5: ffe5445ac39f30efc397899351d12b07 - Dumped twice with the same result (MD5 match).
FW: Jailbreak v6.6

SmokeMonsterPacks, 2018-03-28:
bss has confirmed this issue with his cart dump as well.

bsstephan, 2018-03-28:
Yup, same behavior, ROM dumped via CopySNES crashed after I won the first match.

For whatever it's worth, playing a Retrode-sourced dump off of my SD2SNES works fine too.

TheRealFlamepanther, 2018-04-07:
Happened with mine too.

SmokeMonsterPacks, 2018-07-26:
I'm going to close reports for game that dump incorrectly. If a game doesn't dump right, you need to adjust the settings manually. There will be games that never dump though, and these will not get fixes I'm told.

closed 2018-07-26 by SmokeMonsterPacks