SmokeMonsterPacks/Super-NT-Jailbreak #94: Super Gameboy optimized scaling option

kylophone, 2018-03-29:
I'd love to see an "overscan" function on the Super NT.  This would be hugely useful for scaling Super Gameboy games up to fill the screen. This would be similar to what folks have been doing with the Framemeister/OSSC.

Bits-n-Stuff, 2018-03-30:
Perhaps the ideal situation here wouldn't be just an overscan option but to have specific Super Gameboy 1 & 2 options? Perhaps something along the lines of these?
Increasing the integer scaling to 7x. (I know it's still outputting it as a SNES game so that may not even be possible.)
Is there a way to remove the stuttering that you get ever few seconds? Perhaps something akin to the frame buffer modes used for SNES games?

SmokeMonsterPacks, 2018-04-05:
Closing to focus on bugs. This issue will still be publicly accessible.

closed 2018-04-05 by SmokeMonsterPacks