SmokeMonsterPacks/Super-NT-Jailbreak #97: Improvement: SPC compatibility

ruiner9, 2018-03-31:
The sound engine of the Super Nt is fantastic, so I'd like to see the ability to play SPC sound files from the SD card browser. Perhaps some sort of per-channel visualizer as well. This would be a great way to turn the console into an SNES jukebox for long-term listening.

cacophony1, 2018-03-31:
Kevtris is planning on adding this SPC player to the official Super Nt firmware soon: https://www.youtube.com/watch?v=Q2WJCgOW4qM

SmokeMonsterPacks, 2018-03-31:
This is from Discord: 

> kevtris - 03/21/2018
yes I want to redo scanlines totally
and add spc playing
though it will most likely play SPC2 which is a format I made
maybe I can make it load normal SPCs too
SPC2 is a way to pack all the subtunes of a game into a single file that does not use RAR
and it can be used by real hardware
it is roughly analogous to NSF then

SmokeMonsterPacks, 2018-03-31:
I'll go ahead and close this since something is in the works.

closed 2018-03-31 by SmokeMonsterPacks