SmokeMonsterPacks/Super-NT-Jailbreak #98: Bug: CopySNES creates wrong + incomplete dump of Mega Man X3

bsstephan, 2018-03-31:
CopySNES dumps Mega Man X3's ROM incorrectly. (For whatever it's worth, it gets X2 right.) Bad ROM:

MEGAMAN X3.smc: 1310720 bytes; MD5SUM 26d885737579e979f46e3439db3ba4bc
Mega Man X3 (USA).sfc (No-Intro): 2097152 bytes; MD5SUM cfe8c11f0dce19e4fa5f3fd75775e47c

At 0x00100000 CopySNES starts reading all 0s, and gives up at 0x00140000 (No-Intro goes through 0x001ffff0).

Drakonas on the Discord has the same behavior and might have more to add.

Drakonas, 2018-03-31:
[image]

The rom is incomplete. Even playing in an emulator, it fails to load.

GreatHierophant, 2018-04-01:
Confirmed on discord that Rockman X3 gives a wrong dump but Rockman X2 gives a correct dump.

SmokeMonsterPacks, 2018-04-01:
You should be able to get games like this to dump by changing the settings. MMX3 has an incorrect header on the actual cart. This doesn't prevent the game from playing, but it does prevent it from dumping correctly. Games with bad or incorrect headers will never dump correctly, no matter what changes are made to the firmware. 

So for games with bad headers (like the Game Genie), you will need to find the correct settings to get them to dump. 

Does anyone know of a site or guide with the correct settings? Basically, a guide to EEPROM burning should be useful to figure out the necessary settings.

SmokeMonsterPacks, 2018-04-01:
On the other hand, this may be a bank-switching problem that can be fixed via fw. Could you guys please play around with the dumping settings and see if you can get a good dump?

SmokeMonsterPacks, 2018-04-01:
After chatting with you guys on Discord, it is sounding like this is a bank-switching problem. Please play with the dump settings and report back if you find a way to successfully manually dump it.

frederic-mahe, 2018-04-02:
As a side note, dumping *Rockman X* (v1.0) might be a problem too. The PCBs were "corrected" (see [byuu's pictures](https://twitter.com/byuu_san/status/972951117041688577)) and the correction affects the way the ROM is read.

SmokeMonsterPacks, 2018-04-02:
A couple of people have dumped Rockman X with an identical hash to No-Intro.

bsstephan, 2018-04-03:
I only had a chance to try a couple things but had no further success.

Edited the detected settings to 4 MB, LOMEM: Got a file 2621440 bytes large (larger than the 2097152 No-Intro ROM) but still starts reading all 0s starting at 0x00100000.
Edited the detecting settings to 4 MB, HIMEM: File's the same size, 2621440 bytes, but all 0s (probably not surprising)

What's interesting is that the cart is detected at 2 MB, but dumping that produces a file 1310720 bytes. Double the detected size, and you get double the produced file, but it's still a file with a weird size alignment. So that's definitely odd, but either way, it looks like CopySNES is having issues and reading all 0s long before the ROM should be "over". Probably the "thinking" is off somehow, but I don't really know how to alter its behavior other than changing the detected settings.

SmokeMonsterPacks, 2018-07-26:
I'm going to close reports for game that dump incorrectly. If a game doesn't dump right, you need to adjust the settings manually. There will be games that never dump though, and these will not get fixes I'm told.

closed 2018-07-26 by SmokeMonsterPacks