SmokeMonsterPacks/Super-NT-Jailbreak #100: Glitches: Harukanaru Augusta 3 Masters New

MrGlz, 2018-04-04:
## Expected behavior and actual behavior

Graphical glitches upon match start and continue throughout game.

[image]

This occurs on Harukanaru Augusta 3 Masters New but most likely also occurs on Pebble Beach no Hotou New: Tournament Edition.

## firmware

- firmware version: 4.5

cacophony1, 2018-04-04:
This same behavior was also reproduced by marvelus10 on discord

G-lithcz, 2018-04-20:
Confirming that this behavior also occurs on Pebble Beach no Hotou New: Tournament Edition.

GreatHierophant, 2018-08-01:
No such issues are present when running these games with the new sd2snes SA-1 core.  The core has some minor graphical issues on the Super Nt and original consoles.

Rahzadan, 2018-08-02:
I can also confirm that the specific graphic anomalies described (and shown) in this bug report are **not present** when using SD2SNES new SA-1 core (v02). This could simply be because the SA-1 implementation on SD2SNES is still buggy and the specific SA1 instructions that cause this issue are simply not implemented yet on SD2SNES. Unfortunately I don't have a real cart of this game to compare with. I will update this report if and when a future SD2SNES SA-1 core revision is released with cause these symptoms to show up, as they do on a real cart.

Rahzadan, 2018-08-19:
Issue still does **not** occur on SD2SNES SA-1 firmware v06 on Super NT fw 4.5 - must be a real-cart-only issue... Very odd that the real cart would have these issues but not SD2SNES.

gingerbeardman, 2019-05-10:
I can confirm this using a real cart of *Pebble Beach no Hatou NEW*

https://imgur.com/a/aWvfwPk

gingerbeardman, 2019-08-10:
Can anybody retest on latest firmware? Thanks!

gingerbeardman, 2021-09-19:
Still an issue in 2021: https://retro-gamer.jp/?p=16131

tjanas, 2023-05-04:
@MrGlz @G-lithcz @gingerbeardman can you please re-test your cartridges with the latest official firmware? https://www.analogue.co/support/super-nt/firmware/5.1

tjanas, 2023-05-05:
Change log: “Fixed Harukanaru Augusta 3 graphics issue.”

gingerbeardman, 2023-05-05:
Pebble Beach NEW Tournament Edition

Confirmed fixed

[image]

[image]

closed 2023-05-05 by tjanas

tjanas, 2023-05-06:
I don’t know. You would have to ask someone from Analogue.