SmokeMonsterPacks/Super-NT-Jailbreak #101: SD2SNES with out its SD card doesn't display "No SD card found!" message

RaspberryAlpine, 2018-04-05:
- Super Nt firmware version: 6.6
-SD2SNES firmware sd2snes_gsu_v01

When powering on the Super Nt with the SD2SNES without the SD2SNES's SD card results in a dark grey screen:
https://cdn.discordapp.com/attachments/418896436428537859/431270693225627649/unknown.png

When powering on my SNES jr with the SD2SNES without the SD2SNES's SD card results in a dark grey screen with the message "No SD card found!":
https://cdn.discordapp.com/attachments/418896436428537859/431270693225627649/unknown.png

Rahzadan, 2018-04-05:
I can confirm the same issue on Super Nt firmware 4.5.

I also tested in my 2/1/3 SNES, and it showed "No SD card found!"

Rahzadan, 2018-04-05:
Also confirming that this issue also occurs on SD2SNES original firmware 0.1.7e

Rahzadan, 2019-08-14:
Issue is still present with Snt firmware v4.9 / JB6.9 and SD2SNES OG (Not sure about SD2SNES Pro)

tjanas, 2023-05-08:
@RaspberryAlpine @Rahzadan can you re-test your sd2snes with [firmware v5.1](https://www.analogue.co/support/super-nt/firmware/5.1)?

Ramsis-SNES, 2023-05-08:
This likely was an sd2snes-related issue which should be fixed since [this commit](https://github.com/mrehkopf/sd2snes/commit/6e221c3917a7eed303619ba0369b0c05cc3189f3) (cf. [here](https://github.com/mrehkopf/sd2snes/issues/186) and [here](https://github.com/mrehkopf/sd2snes/issues/201)). I'll try once again on v5.1 once I get home to make sure, and edit this post accordingly.

Rahzadan, 2023-05-09:
Confirmed working normally on Snt 5.1 & SD2SNES 1.11.0.
As @Ramsis-SNES said, it looks like this was an issue with SD2SNES/FXPAK fw (which has since been fixed) rather than SuperNT.

closed 2023-05-09 by tjanas