SmokeMonsterPacks/Super-NT-Jailbreak #104: Super Star Wars randomly pauses in stage 2 (landspeeder part)

cacophony1, 2018-04-17:
I've seen Super Star Wars randomly pause a number of different times while playing the second stage (landspeeder level). It seems pretty random but the first time I played that level it occurred more than once, and I was able to reproduce the issue again on a subsequent play.  But it is intermittent so may take a few attempts.

To be clear I'm only using the directional pad and Y and B buttons, but the game will act like I just hit Start (pause). At this point the game will remained paused until I hit start, at which point I can resume play.

## ROM
This is using the USA Super Star Wars rom from a recent Smokemonster SNES Everdrive set (v5.2)

## firmware
JB 6.6

FWIW, rezb1t on discord had this to say in response:
"Huh. Interesting, that happened to me in a previous firmware
thought it was fixed but I didn't test the speeder level too thoroughly
just like 2 playthroughs and didn't see it"

frederic-mahe, 2018-04-17:
Thanks @cacophony1 

Assuming you are referring to *Super Star Wars: A New Hope*:
```
Super Star Wars (USA).sfc
sha256: c6bd7239cb2074ff1c471d14535bead6290bba2d7c75b4f03209cb114877b4c8
sha1:   dbff9bbb155f09376c4c7fe3d68e77c4a444ae3c
md5:    ea840d3722f90dee824f5064e171e510
crc32:  acbcae7c
```
This is indeed a valid dump, confirmed by [byuu](https://preservation.byuu.org/games/Lj4b/l7teQFeO).

cacophony1, 2018-04-17:
On Discord, Masta_Fu said:

> Yeah, both cart and rom i have the same problem
But with me i didn't have to hit start to resume game
rom version says 1.1, not sure what my cart is
seems to be very rare and very random

And then -Glitchz said:

>  It seems to be that the up directional button is triggering start somehow. If you press start on that stage and then hold up it will unpause the game too. On other stages holding up after pressing start does nothing.

cacophony1, 2018-04-17:
I also noticed that trying to pause via the start button on the 3rd (?) stage (inside the landspeeder) was kind of difficult, as if it was getting the key twice and then leaving me in an unpaused state. I'll try to investigate this a bit more later...

GreatHierophant, 2018-04-18:
A tell-tale sign of this behavior is if pressing the pause button firmly does a quick pause/unpause.  You can get a pause to take with a light press of the start button but pressing Up on the D-pad for a second or two will unpause the game during the landspeeder stages.  This behavior also occurs on the snowspeeder stages in SESB.

G-lithcz, 2018-04-20:
The light pause/unpause action indicated above by GreatHierophant also occurs on real hardware (2/1/3).  What does not occur on real hardware is the unsolicited pauses that randomly occur during the landspeeder stages of all three versions of Super Star Wars. Super Star Wars - stage 2 / Empire Strikes Back - use password RCWJMF / Return of the Jedi - stage 1. One other action that does not occur on real hardware is the ability to unpause during the landspeeder stages by pressing up.  The pause button definitely seems off even on real hardware but these two issues do not occur.

frederic-mahe, 2019-08-09:
The newly published jailbreak firmware should fix that issue. Could you please confirm @cacophony1 ?

lenscritic, 2019-08-10:
I tried through Super Star Wars stage 2 twice and didn't experience any unexpected pauses. So it seems fixed to me.

Thank you mysterious JB author! :D

cacophony1, 2019-08-10:
I played through Super Star Wars stage 2 twice and didn't experience any unexpected pauses. So it seems fixed to me.

Thank you mysterious JB author! :D

closed 2019-08-10 by cacophony1