SmokeMonsterPacks/Super-NT-Jailbreak #105: Magical Pop'n (J) Translated ROM locks up after beating rock boss in Stage 3

siegarettes, 2018-04-17:
using Aeon Genesis translation patch with original Japanese ROM. Gameplay locks up after defeating the rock boss in stage 3 (https://youtu.be/8zgMAandcX4?t=28m41s). Defeated it with bomb item if that makes a difference. 

Game locks up and freezes gameplay, but the boss theme audio can still be heard. 

- firmware version: JB 6.6

GreatHierophant, 2018-04-18:
User -Glitchz on discord had no issues getting past the boss with the original untranslated ROM.  Could you try the translated game on original hardware?  The Aeon Genesis translation is from 2003 and could be buggy,

siegarettes, 2018-04-18:
Unfortunately I don't have access to original hardware or a way to load it.
I'll try it again to see if I can replicate the bug.