SmokeMonsterPacks/Super-NT-Jailbreak #106: Graphical glitch in Seiken Densetsu 2 cave when ballooned by bat

mantixero, 2018-04-19:
## Expected behavior and actual behavior
Graphics glitch when being attacked by bats with a balloon in the cave by the Dwarf Village where you first meet Popoi. It happened while playing with 2 players simultaneously. Did not try it with 1 player only.

## Steps to reproduce the problem
Play with 2 players in the cave with Popoi and Randy and get ballooned by a bat.

## ROM name and hash value
No-Intro Name: Seiken Densetsu 2 (Japan)
(No-Intro version  20130701-030720)
ROM/File SHA-1: B78A9A844D165345631CEA1B5246C8FCBCDBC162

## firmware

- firmware version: JB 6.6

## relevant Super Nt settings
480p
Zero lag mode
Interpolation disabled

mantixero, 2018-04-20:
The issue is also present on real SNES hardware as confirmed in a video by seadona#1221 in the Discord chat, so it isn't specific to the Super Nt.

closed 2018-04-20 by mantixero

SmokeMonsterPacks, 2018-04-20:
Thanks for looking in to it, @mantixero mantixero :D