SmokeMonsterPacks/Super-NT-Jailbreak #107: CopySNES fails for Far East Of Eden - Tengai Makyou Zero (JP)

DeadlySystem, 2018-04-26:
## Expected behavior and actual behavior

I am trying to use CopySNES of JB6.6 to dump the japanese game "Far East Of Eden - Tengai Makyou Zero" (see https://superfamicom.org/info/far-east-of-eden-tengai-makyou-zero). CopySNES detects it as LoROM, but according to superfamicom.org it's HiROM. Regardless of the setting, the dump comes out wrong - also with ExLoROM, ExHiROM and Flat settings. In all cases, I obtain a 5 MB file (which is the correct size). The first 2 MB of this file are correct; after that, the first MB is repeated three times, i.e. the dump has the structure ABAAA where every letter represents 1 MB worth of data.

Note that this game is using the SPC7110 chip as well as RTC functionality. If I set the chip to None and to a 16 MB flat dump, I still cannot find any of the data contained in a correct ROM dump after the 2 MB boundary.

## Steps to reproduce the problem

Use CopySNES with a regular copy of SHVC-AZRJ-JPN (Far East Of Eden - Tengai Makyou Zero). The dump comes out wrong.

## ROM name and hash value

The ROM dump produced by CopySNES has MD5 790118A30BD053F6BFE9B10718CBD37F.
The expected MD5 hash is 29998124A67DEDA111D5ACA8AAF5DCA5.

## firmware

- firmware version: JB6.6

## relevant Super Nt settings

Default settings as after flashing JB6.6.

alexcom84, 2018-04-27:
I can confirm the copy snes bug with Tengai Makyou Zero, original SHVC-AZRJ-JPN. 
The english repro instead is getting only 5mb dump instead of the expanded 7mb. (understandable with the repro, given the complicated procedure to make it work)

SmokeMonsterPacks, 2018-07-26:
I'm going to close reports for game that dump incorrectly. If a game doesn't dump right, you need to adjust the settings manually. There will be games that never dump though, and these will not get fixes I'm told.

closed 2018-07-26 by SmokeMonsterPacks