SmokeMonsterPacks/Super-NT-Jailbreak #109: CopySNES creates wrong + incomplete dump of Street Fighter Zero 2

alexcom84, 2018-04-27:
Copy SNES is dumping only 3mb of Street Fighter Zero 2 instead of 4mb. Tried dumping Start Ocean and I get a valid ROM, so it's not SSD-1 the issue.

Copy SNES settings:
ROM 4mb
ExLoROM
RAM 0

davidpgil, 2018-05-13:
I can also add that I have the Japanese Super Famicom cartridge for this game and it doesn't run either. However, SFZ 2 does work with the official Analogue firmware.

(By the way, so far the jailbreak firmware has a lot better compatibility for me than the official one.)

ShinKaz, 2018-06-16:
> I can also add that I have the Japanese Super Famicom cartridge for this game and it doesn't run >either. However, SFZ 2 does work with the official Analogue firmware.

>(By the way, so far the jailbreak firmware has a lot better compatibility for me than the official one.)

Yeah, no.

https://imgur.com/a/9JAiy6q

SmokeMonsterPacks, 2018-06-16:
So this works?

ShinKaz, 2018-06-17:
the car works perfectly fine on my end on JB FW 6.6. (both pal and ntsc-j versions, I don't own the ntsc-u version to test)

as for dumping, I suppose this is still unsolved, I didn't try.

as a side note, I'm pretty sureit is possible to make a "decompressed" rom version of this game and run it on the super nt, like Star Ocean. (you can find the decompressed assets online, I just don't have to knoledge to build the decompressed rom)

davidpgil, 2018-07-26:
@ShinKaz @SmokeMonsterPacks what @Ghost meant is that the dump doesn't work. I have the Jpanaese cart too and it works fine. Its just that the CopyNES dump doesn't work.

SmokeMonsterPacks, 2018-07-26:
I'm going to close reports for game that dump incorrectly. If a game doesn't dump right, you need to adjust the settings manually. There will be games that never dump though, and these will not get fixes I'm told.

tjanas, 2023-07-11:
@alexcom84 @ShinKaz @0xvividmirage CopySNES issue with dumping S-DD1 carts has been reported as fixed, please retest with [JB v7.2](https://github.com/analoguejb/Analogue-Super-Nt-JB).

alexcom84, 2023-07-11:
> @alexcom84 @ShinKaz @0xvividmirage CopySNES issue with dumping S-DD1 carts has been reported as fixed, please retest with [JB v7.2](https://github.com/analoguejb/Analogue-Super-Nt-JB).

Yup, it’s fixed in latest fw!

closed 2023-07-11 by alexcom84