SmokeMonsterPacks/Super-NT-Jailbreak #110: Single line tear on output with interpolation enabled for non-integer scale

solzun, 2018-04-28:
When setting the SuperNT to a width/height setting of 1365, 38, 1149, 42 (recommended settings from the Digital Foundry and My Life In Gaming Super NT reviews on YouTube) and enabling both horizontal and vertical interpolation under the scaler menu, there is a single line of tear introduced near the top 25% of the screen.

Steps to reproduce:
- Load up a USA/NTSC version of Chrono Trigger, start a new game, and when the camera scrolls diagonally  from the town down towards the main house you can see it tear along the castle/forest. It's noticeable as well in NTSC version of EarthBound in the intro part where the camera scrolls across the city at night, or ShadowRun when you escape the morgue at the beginning.

Video posted here (though it's pretty hard to see in the recording, easier to see from a distance on a HD/UHD TV): https://www.youtube.com/watch?v=cR3Pu619qNs

Disabling interpolation and moving to an integer scale seems to fix the problem: 1536, 38, 1200, 42

Firmware:
JB 6.6

Super NT settings:
- 1080p60
- No scaler/scanlines enabled (though it doesn't matter)
- Vertical and horizontal interpolation is **not** disabled.
- Width/Height settings: 1365, 38, 1149, 42

tjanas, 2021-05-08:
@solzun is this issue still outstanding in the latest JB version?

tjanas, 2023-07-12:
Using 1411 width, 1158 height, 66 hpos, 26 vpos, I do not see the glitch shown in your video with JB v7.2.

closed 2023-07-12 by tjanas