SmokeMonsterPacks/Super-NT-Jailbreak #111: Screen flicker on Uncharted Waters

vinylspinner71, 2018-05-01:
So this is with an North American Cart

First  .....Firmware is 6.6
screen is in 1090x60
width =1792
height 1080
no scalers
fully buffered

So this is kinda hard to explain so bear with me

When you are sailing around the word , the screen (sailing map) will shift ...to help create movement.
Now the screen will shift up and down

This only happens when the screen is shifted up towards top of tv

The top 20% of the screen will flicker with horz lines.  This will happen when a new screen appears sailing east to west or west to east, and in the "up position" on the tv

if the screen is on the "down" position, no flickering

Sailing north south , it will only happen when the screen is moving from ""down" to "up", the ships position does not matter .

I dont have a SNES to test it against.

frederic-mahe, 2018-05-01:
@vinylspinner71 I assume you are talking about *Uncharted Waters* (SNS-QK-USA), not about *Uncharted Waters - New Horizons* (SNS-QL-USA).

G-lithcz, 2018-05-01:
I tested both Uncharted Waters and Uncharted Waters - New Horizons and only noticed the screen shift effect in Uncharted Waters. I did not notice any differences from the SNT and an original 2/1/3 in the top 20 percent of the screen when sailing around.  The top does flicker occasionally but nothing inconsistent with real hardware. Unless others find something I'd recommend closing this out.

vinylspinner71, 2018-05-01:
Yes on the first Uncharted Waters,  I am going to roll back the Super NT to a older firmware tonight.  It does seem worse with the new 6.6 but that can be me only thinking that way.  I have two Super NT's and will test two firmware versions.

vinylspinner71, 2018-05-03:
Its hard to tell a difference between firmware's......but there is a difference between Super Nt's (Purple and Black model) running the same firmware.  

Both have screen flicker but one is FAR more often.  

The one (black SNT) also has a glitch...everytime you start the system it has a quick screen flicker 40-50 seconds after starting it and never does it again.  I am in contact with analogue about it

So yes to screen flicker, black model was more than the SNES purple model ...hard to tell difference on firmware...and one (black model) maybe hardware problem related.

frederic-mahe, 2018-05-03:
That's puzzling... @vinylspinner71 just to be extra-sure, did you clean the pins of your cartridge? A bad contact could explain why you get different results when you plug the cartridge to another system (assuming the color-variants of the Super NT are all internally identical).

vinylspinner71, 2018-05-03:
Contacts have been cleaned with dexoit and followed up with isopropyl alcohol.   
The glitch on the black model has happened from day one across all games, rom and cart.

It only happens when you first turn it on and start playing 40-50 sec in.  and it only happens once....  and does not again unless you physically power down and restart.

So back to this....I agree with G-lithcz, since it looks like by his testing , this occurs on the SNES hardware, this should be closed unless anyone else try's it and comes up with something different.

closed 2018-05-03 by SmokeMonsterPacks