SmokeMonsterPacks/Super-NT-Jailbreak #113: Audio skipping after saving

sherzinger, 2018-05-08:
Just got a Super NT today and flashed it with the jailbreak v6.6.
I notice that after saving and restarting the console the audio starts skipping. Every couple of seconds I completely loose sound for 2 seconds.

Im playing SMW romhacks, but the issue carries over to other games aswell. A factory reset of the settings solves the issue until the next save.

Any idea what could cause this?

SmokeMonsterPacks, 2018-05-08:
This is some kind of problem with your TV.

closed 2018-05-08 by SmokeMonsterPacks

sherzinger, 2018-05-15:
@SmokeMonsterPacks I had some time to test this now and I'm fairly certain that the Super NT handles HDMI very poorly. I get frequent audio skips that only disappear when reconnecting the cable. Also it entirely fails to wake up my sreen. I have to manualy turn it on before starting the console.

I have none of these issues with any other system here.