SmokeMonsterPacks/Super-NT-Jailbreak #114: Glitches: Rockman X3 sprites disappearing

mantixero, 2018-05-10:
## Expected behavior and actual behavior
Sprites should stay fully visible on screen, but often have portions missing in certain situations.

## Steps to reproduce the problem
Jumping out of a Ride Armor is an easy way to reproduce the issue. Getting hit by an enemy and activating X's shield (once you get the shield upgrade) also tends to show portions of the shield missing.

## Firmware
- firmware version: JB6.6

## Relevant Super Nt settings
I do not have 64-sprite mode enabled.
[image]
[image]

frederic-mahe, 2018-05-10:
@mantixero I've tested on my 1-chip Super Famicom with an original cartridge and I did not see sprite clipping when shield activates. Can someone else please confirm that there is no clipping on original hardware (ideally, tests should be done on a 2/1/3 NTSC model)?

G-lithcz, 2018-05-11:
This issue was replicated on a 2/1/3 using an original Rockman X3 cart. Just another example of how faithful the Super NT truly is. In order to avoid this on the SNT just enable the 64 sprite tiles setting.

closed 2018-05-11 by SmokeMonsterPacks