SmokeMonsterPacks/Super-NT-Jailbreak #122: Super Metroid: ROM hangs on Mother Brain explosion

chappg, 2018-07-24:
##TITLE: Super Metroid: ROM hangs on Mother Brain explosion

## Expected behavior and actual behavior
Expected Behavior: 
At 1:10 in the following video, Mother Brain is destroyed in her glass jar.  The screen centers and a set of explosions occurs as Mother Brain changes to her second form:
https://www.youtube.com/watch?v=_mGOFQVhtFs 

Actual Behavior:
At 1:15 in my video, Mother Brain is destroyed in her glass jar. The screen centers, but the ROM hangs before explosions occur: https://www.youtube.com/watch?v=pdpULctE8x0 

Note: The Super NT system menu is still responsive during the hang.

## Steps to reproduce the problem
1. Defeat Mother Brain in her glass jar
2. Watch as the screen re-centers, and the system hangs before the explosions occur

## ROM name and hash value
12b77c4bc9c1832cee8881244659065ee1d84c70c3d29e6eaf92e6798cc2ca72	SD2SNES/1 US - S/Super Metroid (Japan, USA) (En,Ja).sfc	da957f0d63d14cb441d215462904c4fa8519c613	21f3e98df4780ee1c667b84e57d88675	d63ed5f8

## firmware

- firmware version: JB6.6

## relevant Super Nt settings

I believe I'm on default settings

NOTES:
I can repro this consistently with the steps described. Unfortunately, reaching Mother Brain requires a full play-through

frederic-mahe, 2018-07-25:
Thanks @chappg for that detailed report. I took the liberty to modify a bit the formatting of the issue. Ideally, we would need someone to reproduce the bug with:

1. the same ROM and settings,
2. with an original US cartridge.

Any Super Metroid speedrunner in the room? :-)

chappg, 2018-07-25:
I don't have a cartridge for testing, but I can find one in an month.

I attempted to repro ten times today, and unfortunately (fortunately?) the crash is not occurring. I saw the crash three times in total, consecutively, including the third time when I was recording the video. I wonder if all this practice has made me play "better" and now the specific (and undocumented) extra condition is difficult for me to repeat.

Studying the video, I noticed that the number of missiles in inventory went to zero at the same moment Mother Brain is destroyed. You can see the game switch over to the default weapon in the extraneous shots as the screen centers. Attempting to isolate this condition hasn't led to another repro of the hang, though.

@frederic-mahe how can I dump the settings for you?

frederic-mahe, 2018-07-25:
@chappg and other contributors in general, please **never** **ever** attach ROM files of official games. GitHub does not provide a way to delete attached files, so I had to directly contact their support, asking for an administrator to delete the file.

On the other hand, attaching save files is encouraged for reproducibility.

frederic-mahe, 2018-07-25:
> how can I dump the settings for you?

@chappg please list your Super Nt settings that are not default settings, if any (you mentioned you were using mostly default settings).

chappg, 2018-07-25:
<closed by accident; re-opening>

juraisa, 2018-08-15:
I tested this on an SD2SNES Rev H running the v05 SA-1 firmware, no crash. Tested by loading ROM via Jailbreak SD Card browsing, also no crash. Cannot simulate reported behavior after 5 attempts. 

NTSC / Launch System Timing / Zero Delay

lukeshumard, 2019-01-07:
Just ran this on an original US cartridge several times, no crash. Went to 0 missiles when ending fight with Mother Brain 1 as well, but couldn't reproduce the issue.

I also used CopySNES to get the ROM, same SHA256 hash as original issue.

**Firmware**
* JB6.6

**Super NT settings**
* 1080p60
* Width 8:7
* Height 4.5x
* No Scaler
* No Scanlines
* Pseudo-Hires Blending
* NTSC
* Launch System Timing
* Zero Delay

closed 2019-01-08 by SmokeMonsterPacks