SmokeMonsterPacks/Super-NT-Jailbreak #123: Plugging Retro Receiver into port 2 while Super Nt powered on causes crash

perilsensitive, 2018-07-25:
I have two 8BitDo Retro Receivers.  If I plug either one of them into the second controller port while the Super Nt is powered on, the Super Nt crashes (TV loses video signal, Super Nt doesn't appear to respond).  This does not happen if I plug a receiver into the first controller port while powered on, nor does it happen when plugging a wired controller into the second port.

As long as the retro receiver is plugged into the second controller port before the Super Nt is turned on, everything works fine.

I don't usually hot-plug controllers anyway, so this isn't a huge concern for me, but I wasn't sure if it was a problem with my hardware or if it is a general issue. 

Super Nt firmware version: JB6.6

Retro Receiver firmware version: 1.30

boongee, 2018-07-25:
> Thanks @perilsensitive <https://github.com/perilsensitive>
>
> [wild speculation] 8BitDo Retro Receivers might draw more power than a
> wired controller, causing a temporary drop that crashes the system.
>
> If confirmed, the Super Nt documentation could be updated to discourage
> hot-plugging.

This is an issue with the current (and previous) official firmwares as
well.   Also, with both receivers plugged in, the console fails to boot and
the receivers flash orange/green.  Quickly power cycling a allows the
console to boot.

chappg, 2018-07-25:
I can repro and have uploaded a video here: http://www.youtube.com/watch?v=QvbifobADF8

Note that the running game doesn't crash, and the HDMI signal can be restored on my unit by unplugging and re-plugging the HDMI cable.

perilsensitive, 2018-07-25:
@boongee:  I actually don't have any issues powering the console on with both receivers plugged in.  Was that true on all of the firmware versions you tested, and were you testing with a cartridge inserted?  I can test that tonight when I get home.

@chappg: I didn't think to try disconnecting and reconnecting the HDMI cable; I'll try that too.

SmokeMonsterPacks, 2018-07-25:
You shouldn't hotplug 8bitdo receivers since they have a higher current draw than standard controllers. What is happening when you plug the second in is that the over-current protection circuit is kicking in and shutting down the console to protect the port. Kev spoke about this somewhere, and it's actually a feature. If it didn't do this, it could damage the SNT.

closed 2018-07-25 by SmokeMonsterPacks

perilsensitive, 2018-07-25:
That's good to know; thanks.