SmokeMonsterPacks/Super-NT-Jailbreak #125: Glitches: Shounen Ashibe - Goma-chan no Yuuenchi Daibouken (Japan)

black-francis, 2018-10-17:
## Expected behavior and actual behaviour

On the right hand side of the title screen a single black pixel can be seen moving rapidly back and forth on either the purple pillar or the grass next to it, the majority of the time it shows in the grass.

This issue cannot be seen on either Snes9x or Higan and a user on Discord has confirmed that they do not see it on real hardware with an SD2SNES. 

## Steps to reproduce the problem

Simply proceed through the developer splash screen and intro to the main title screen.

## ROM name and hash value

Shounen Ashibe - Goma-chan no Yuuenchi Daibouken (Japan) 

ROM MD5:   AC3523F97A0A4567A6E3CCFE1EC47BA0

## firmware

Latest Jailbreak.

## relevant Super Nt settings

Issue is present regardless of settings tested.

frederic-mahe, 2018-10-17:
For reference, here are the hash values of the ROM:
```
SHA256: 5e995adb1a69ff9d9ea43697de7665619abb6aee13eef8fba059d9a197b6dd0c
SHA1: 078c359681a39c0e2b8261f053ea9e85f68d5202
MD5: ac3523f97a0a4567a6e3ccfe1ec47ba0
CRC32: d01ab6a8
```

frederic-mahe, 2018-10-29:
@black-francis I see the same flickering pixel when playing on a Super Famicom 1CHIP-1 with a SD2SNES.

closed 2018-10-29 by SmokeMonsterPacks