SmokeMonsterPacks/Super-NT-Jailbreak #126: Glitches: Wild Guns (USA)

black-francis, 2018-10-18:
## Expected behaviour and actual behaviour

During the in game shooting sections a single black pixel can be seen moving rapidly back and forth on the lower right hand size of the playable area, just above the wooden bezel which holds the players lives, time remaining etc. 

Issue not present on either Higan or Snes9x but still needs testing on real hardware.

## ROM name and hash value

Wild Guns (USA)
                         
ROM MD5 94D30063D9B17567D32079E766251569        
ROM SHA-1 1ECF28D5F3D2E84E3CC58ED621CB0FC0DD9D70D0
ROM CRC32 085D8D14                                

## firmware

 snt_firmware_verJB6.6

## relevant Super Nt settings

Issue is seen with all tested settings.

frederic-mahe, 2018-10-20:
could be linked to #125 ?

frederic-mahe, 2018-10-29:
@black-francis I see the same flickering pixel when playing on a Super Famicom 1CHIP-1 with a SD2SNES.

closed 2018-10-29 by SmokeMonsterPacks