SmokeMonsterPacks/Super-NT-Jailbreak #127: Glitches: Cropping of image.

black-francis, 2018-10-18:
## Expected behaviour and actual behaviour
​
Expected: The ability to display graphics to the entirety of the vertical resolution of a 1080p screen. With a 5x integer, stretching the active graphics to the full resolution of your screen or using the vertical position option to push the graphics to the top of the screen should not result in any of the image not being drawn.​
​
Actual: The Super NT is seemingly incapable of using the very top of any display so you get a varying amount of blank black pixels where there should be image displayed. This is most easily noticed when using a 5x integer scale as this results in an entire line of graphics not showing at the top of the screen, resulting in 9 total vertical pixels being clipped instead of the expected 8 pixels (4 off the top and 4 off the bottom). 
​
## Steps to reproduce the problem​
​
The problem is present regardless of the video settings you choose or the resolution selected but these settings show it clearly. The only reason you cannot show this with a 4x integer is that you can not move the active graphics high enough up the screen with the vertical position option, but I have no doubt you would see the same issue if you could.​
​
**1080p 5x integer**: Screenshot below and text created by Firebrandx and used with his permission.​
​
[image]

"If you max the vertical out to 5x (1200), the visible window only shows 1175 vertical res, resulting in a total clip of 9 pixels instead of just 8. I cross-referenced this on two different displays as well as my E1S capture card, and the result was exactly the same. Here's the lossless capture showing the net loss of an extra pixel row" "Note how also in that image the vertical position had to be lowered to 40 in order to center in on the active 1175 lines."​
​
**1080p non integer:** Screenshot below and text created by Firebrandx and used with his permission.​
​
"It gets even worse for non-integer scaling. If you enable HV interpolation and attempt to scale to fit the screen vertically, the math dictates you should be able to set the res at 1410x1157. However, due to the faulty scaling algorithm, this ends up clipping a couple rows of pixels. The closest you can get is 1406x1153, and you have to set the vertical position to 44. Below is the lossless capture for proof of that as well"​
​
[image]

​"For those that like fullscreen or clipped overscan 5x, you're not able to get the best experience in those modes."​
​
**720p 3x integer and 480p 2x integer:** Simply move the image towards the top of your screen with the vertical position option and you will see a large amount of pixels are blanked out at the top of the screen when they shouldn't be.​
​
## firmware​
​
snt_firmware_verJB6.6

SegaSnatcher, 2019-06-23:
Kevtris has confirmed that he's back porting the fix he did for Mega SG to Super NT.  So all 1080 lines will be visible.

frederic-mahe, 2019-08-09:
The newly published jailbreak firmware should fix that issue. Could you please confirm @black-francis ? Thanks.

black-francis, 2019-08-09:
unfortunately the issue is still present with the latest jailbreak firmware. there is also another issue where video settings are saving but not applying when restarting the console unless you go into the video settings and press a button, then they take effect.

re4mat, 2021-02-24:
On official 5.0/JB7.0, on my 1080p TV, I'm able to stretch the image all the way to the top and bottom edges when I set my resolution to 1080p and set height to 1150 with a vertical position of 24. On 4.9/JB6.9, I could achieve the same by setting the height to 1140.

closed 2021-02-24 by SmokeMonsterPacks