SmokeMonsterPacks/Super-NT-Jailbreak #130: California Games 2 - Title screen graphic corruption.

welsby, 2018-11-18:
## Expected behavior and actual behavior

When loading the ROM the title screen should be displayed.

On occasion, the top left of the title screen is showing corrupted graphics, sometimes (but rarely) this corruption continues onto other screens. The corruption is similar to that caused by dirty pins.

## Steps to reproduce the problem

No clear steps, happened on first load. Rebooted the system and it went away, then reloaded the same ROM again via the menu and it occured a second time. Out of the 6 times I've tried, it's been corrupted on 4.

## ROM name and hash value

California Games II (U).smc 
19f4a1f08aa55ff3cc8ff7ae60ffe03f0e436e8d8901455f1311f2276497a492

(SHA matches Byuu.org)

Did not occur at all when using the cart on stock firmware.

## firmware

Latest (JB6.6)

## relevant Super Nt settings

Nothing fancy, all default except for some custom scan-line settings. Displayed at 5X.

impetus-maximus, 2018-12-29:
i just tried this ROM with JB6.6. i have no corruption whatsoever on the menu screen. 
tried restarting 10+ times.
micro SD card used; Samsung 8GB Class ➅

frederic-mahe, 2018-12-31:
thanks @impetus-maximus 

@welsby given that the glitch cannot be replicated, I am going to close that issue for now. Thanks again for taking the time to report.

closed 2018-12-31 by frederic-mahe

impetus-maximus, 2018-12-31:
glad i could assist. :)
appreciate the time and effort put into the jailbreak.
cheers