SmokeMonsterPacks/Super-NT-Jailbreak #132: Feature Request: Jailbreak Firmware Update v4.6

itsthedoc, 2018-12-14:
Hello,

The new 4.6 official firmware has been released. The main difference in the release notes is that this supports the Ghostly limited edition.

Will there be any announcements towards merging these changes into the jailbreak?

Thanks in advance.

xerotyme, 2018-12-16:
I just received my Ghostly edition and was wondering the same thing. I don’t want to attempt to jailbreak it and screw it up seeing as how the stock firmware is different from all other Analogues...

itsthedoc, 2018-12-17:
> I just received my Ghostly edition and was wondering the same thing. I don’t want to attempt to jailbreak it and screw it up seeing as how the stock firmware is different from all other Analogues...

I flashed the jailbreak and it works the same as non-Ghostly's, although without the Ghostly custom sounds and graphics. 

In the official release notes, 4.6 works for both Ghostly and non-Ghostly units, with the only difference being Ghostly units will have all customization back after flash. 

You can flash back to stock with no risk of losing the Ghostly features. Hope that helps.

frederic-mahe, 2018-12-17:
Thanks @itsthedoc ! I've [modified](https://github.com/SmokeMonsterPacks/Super-NT-Jailbreak/commit/d7272176fd8df67cdc822eefc69853998dcef54c) the README file to indicate that the current jailbreak firmware also works with Super Nt ghostly models.

closed 2018-12-17 by frederic-mahe

hellodeibu, 2018-12-18:
@frederic-mahe though this doesn't technically solve this issue, as people with Ghostly models who choose to run custom firmware will still not have the custom animations. Wouldn't it still require the custom firmware to be updated with 4.6 support for that reason?

xerotyme, 2018-12-18:
@hellodeibu although you will lose the custom boot sounds and animations while on the jailbroken firmware, it does indeed work as @ itsthedoc has stated. I wasn’t willing to take the initial risk of installing it seeing as how these consoles cost a bit of money and are very limited in quantity. It sucks losing the features that make it unique however, you can always do what I plan to do. I intend to pick up a SD2SNES (due to the fact it has the highest compatibility rate with ROMS) and then restore firmware 4.6 to get the boot sounds and animations back. I hate the fact that it costs over $200 but in this case you really do get what you pay for! You can even get one that matches the console. :) Not to mention the fact it will also exponentially increase the resale value down the line should you ever choose to sell it.