SmokeMonsterPacks/Super-NT-Jailbreak #134: Extend CopySNES feature set (saves, patches, chip passthrough)

therourke, 2019-03-18:
Thanks for the hard work on the jailbreak! I have a couple of feature requests, potentially impossible, but why not ask:

- Ability to load a .sav file from SD whilst playing a cart game OR to transf .sav backups directly to carts.
- Ability to load patches for cart games (would allow loading of translation hacks for instance, similar to the options [offered by the Retro 5](https://gotgame.com/2014/09/04/latest-retron-5-beta-firmware-update-adds-fan-translation-patch-support/) and [Super UFO Pro 8](https://www.youtube.com/watch?v=32NhzMgeN5A).
- Ability to pass through cart chip features such as SuperFX to ROMs loaded from SD card (wishful thinking? Plug in a Yoshi's Island cart, but play a Winter Gold rom? Possible?)

frederic-mahe, 2019-03-18:
Interesting suggestions @therourke 

I'll close the issue to avoid clutter, but don't be afraid, all feature requests remain permanently accessible using label filters.

closed 2019-03-18 by frederic-mahe

therourke, 2019-04-29:
Another feature which would be useful is to change default location of save games, or to save in .srm format (same as .sav, just different extension).

This would be useful for using same SD card in multiple devices