SmokeMonsterPacks/Super-NT-Jailbreak #137: Dumping/Ripping Super Metroid - Creates bad dump (over dumping)

amoore2600, 2019-06-09:
Dumping Super Metroid produces a bad dump with JB6.6 2018-03-27

Dumped checksum:
CRC32 fc725bb5

Known Good checksum:
CRC32 D63ED5F8

I suspect this is do to the strange size of the Super Metorid Rom.  Dumping groups back in the day had issues with dumping Super Metroid too.  I have dumped over +100 games without issue on JB6.6 until I tried  Super Metroid or maybe I just have a bad copy.

amoore2600, 2019-06-09:
I believe that the Super NT over dumps Super Metroid 

**Bad dump from SuperNT (note the files size):**
cksum Super Metroid.smc
2421566781 **4194304** Super Metroid.smc
md5sum Super Metroid.smc 
a5d5ffb7577588263ab1317199867e56  Super Metroid.smc

**Good Known dump:**
cksum Super Metroid (Japan, USA) (En,Ja).sfc
1920880510 **3145728** Super Metroid (Japan, USA) (En,Ja).sfc
md5sum Super Metroid (Japan, USA) (En,Ja).sfc
21f3e98df4780ee1c667b84e57d88675  Super Metroid (Japan, USA) (En,Ja).sfc
http://dorando.emuverse.com/html/super-metroid.html

frederic-mahe, 2019-06-09:
Hum, the Super Nt seems to dump 32 Mb instead of 24 Mb. If you truncate the overdump back to 24 Mb, do you get the expected checksum?
```sh
# if you are on a Unix/Linux system
truncate -s 3145728 old.sfc new.sfc
md5sum new.sfc
```

amoore2600, 2019-06-09:
I just confirmed, removing the last bit gives the correct checksum. I removed everything after offset 00FFFF0.

mike-ost, 2019-09-18:
This issue seems to be fixed in JB6.9 2019-08-09

I just dumped Super Metroid and got:

CRC32: D63ED5F8
MD5 Checksum: 21F3E98DF4780EE1C667B84E57D88675

frederic-mahe, 2019-09-19:
Thanks for reporting that @mike-ost

closed 2019-09-19 by frederic-mahe