SmokeMonsterPacks/Super-NT-Jailbreak #138: new jailbreak for firmware 4.7

micropixel70, 2019-06-24:
When will the jailbreak be released for the new 4.7 firmware? Thanks

frederic-mahe, 2019-06-24:
> v4.7 notes:
> - fixes gameboy audio bug

micropixel70, 2019-06-24:
I have the 4.7 firmware can I jailbreak?  thank you

projectzero, 2019-07-27:
The bug that 4.7 fixes was introduced in 4.6. 4.6 introduced the Ghostly Limited Edition firmware. So for most users, 4.5 (which the latest jailbreak uses) is all you need. 

Yes, you can update if you're already on 4.7, just follow the directions on the main page. As stated, you'll be on 4.5 after the jailbreak but the only users that will impact are those with the Ghostly Edition (you won't have the special boot screens, just the standard versions).

frederic-mahe, 2019-08-09:
The newly released jailbreak firmware includes fixes from all previous official firmware.

closed 2019-08-09 by frederic-mahe