SmokeMonsterPacks/Super-NT-Jailbreak #140: New Updated Jailbreak with last firmware 4.9 fixes

mlassii, 2019-08-08:
Hi! We will see an updated version of the Jailbreak with last fixes from the recently release 4.9 version ?

Thanks!!

v4.9 notes:

- Added DAC compatibility (required for DAC)
- Added user font loading
- Added RGB sliders for standby color
- Added SPC player
- Fixed top missing scanline bug
- Added new scanline system
- L/R audio input bug fix
- Fixed "you cannot use a mouse with this game" on Panic Bomber World
- Fixed Star Wars pausing bug
- Many small fixes and changes

BooBerry, 2019-08-09:
Looks like you'll get your wish, according to Smokemonster's stream announcement: https://www.youtube.com/watch?v=eAVpVuguuyk

> includes Super Nt special chip support

Includes special chip support too apparently. 💯

frederic-mahe, 2019-08-09:
The new firmware is now live.

closed 2019-08-09 by frederic-mahe