SmokeMonsterPacks/Super-NT-Jailbreak #141: Glitches: Mega Man x2 Slowdown

Chronokeys, 2019-08-10:
**Behavior:** 
There's significant slowdown on the main menu and some other areas of the game that is not present in real cart.

**Steps to Reproduce:**
On the main menu scroll up/down through the options to find its very unresponsive, some inputs will not get recognized at all, when choosing an option megaman fires a shot and the screen fades out, this also goes slower. The fight against the sword miniboss also has some slowdown. I've compared this to an actual Rockman x2 cart and you can tell the difference right away.

Firmware: 
JB 6.9

Rom: 
Mega Man X2 (USA).sfc

md5: 
d41d8cd98f00b204e9800998ecf8427e

Stock SNT settings

Nigoli, 2019-08-11:
I played through the Japanese version had had the problem too.

The late prototype also shares these symptoms. Crystal Snail and the Stage Select as you beat more bosses makes it the most noticeable.

I think that the Cx4 is different than X3’s because that is barely noticeable. X3 was questionable to me with fighting the Sigma Virus and maybe when I fought the wireframe mini boss in Blast Hornet’s stage using the Cross Charge shot. I think X3 needs more testing comparing it’s orignal cartridge.

benisjamino, 2019-08-17:
I just get the error message, "cannot open dsp file," on Mega Man X2 and X3

rtrind, 2019-10-03:
> I just get the error message, "cannot open dsp file," on Mega Man X2 and X3

You are probably just missing the special files on the SD card to run DSP roms. Not the error reported on this thread.

re4mat, 2021-02-24:
Was not able to reproduce the issue on JB7.0.

tjanas, 2021-02-24:
Please reopen per @re4mat and @Chronokeys

Chronokeys, 2021-02-24:
Added new info to the OP, the difference should be easier to pinpoint.

SmokeMonsterPacks, 2021-02-24:
@re4mat reported it's fixed in 7.0. OP is tested with 6.9.

re4mat, 2021-02-24:
OP and I had a conversation in Discord after the issue was closed and we determined that the issue is actually still present in 7.0. New details were just added to the original post by the OP.

Nigoli, 2021-02-24:
Have you guys actually tried **playing through the game** to see if the slowdowns that happened to me still exist? It's not just something that happens on the title screen and choosing a selection. It's all the same irregardless of which region.

Even X3 isn't as accurate as a real cartridge here.

tjanas, 2023-07-11:
@Chronokeys @Nigoli please retest with [JB v7.2](https://github.com/analoguejb/Analogue-Super-Nt-JB). It has been reported as fixed in the change log.

closed 2023-07-11 by tjanas

Chronokeys, 2023-07-12:
It is not fixed, dunno what the guy testing this is thinking or why they claim once again that the issue is fixed, it’s clearly not. Guess it must be a tough one to crack since this the 2nd attempt, 3rd if you count first release.

tjanas, 2023-07-12:
@Chronokeys if the change log is not correct and this is not fixed, please open an issue at the new GitHub repo for visibility.

tjanas, 2023-07-12:
https://github.com/analoguejb/Analogue-Super-Nt-JB/issues

Chronokeys, 2023-07-13:
I apologize, upon further testing it appears the game now runs at the correct speed, there is still some slowdown but apparently rock man x2 and mega man x2 both run the wireframe animation at different speed for some reason, so that's that, issue fixed