SmokeMonsterPacks/Super-NT-Jailbreak #142: Glitches: Mario to Wario sky graphic glitch

Chronokeys, 2019-08-10:
## Behavior:
Top half of the sky of the main menu changes color when choosing a different mouse speed.

[image]

## Expected Behavior:
Sky must remain unchanged
[image]

## Steps to reproduce 
Very easy but requires an snes mouse. On the menu simply keep choosing a different mouse speed, low or high, and the sky will change its color. Dont know if it's relevant but this was encountered using a hyperkin snes mouse.

Firmware:
JB 6.9

Rom:
Mario to Wario (Japan).sfc

md5:
d41d8cd98f00b204e9800998ecf8427e

Stock SNT settings

tjanas, 2019-08-11:
Does this happen with version 6.7 of the firmware?

tjanas, 2019-08-12:
FYI I was unable to recreate this with a real SNES mouse and JB 6.7.

Chronokeys, 2019-08-12:
This happened to me on JB 6.7 also, weird, maybe the hyperkin mouse is to blame?

frederic-mahe, 2019-08-12:
@Chronokeys thanks for reporting that issue.

Could you please double-check the MD5 of your ROM? *Mario to Wario* (Japan) (En)'s MD5 should be B1774F03700DFC311BFD86411DB629C3 (see [no-intro](https://datomatic.no-intro.org/?page=show_record&s=49&n=1563)).

d41d8cd98f00b204e9800998ecf8427e is the MD5 of *Taekwon-Do* (Korea).

Chronokeys, 2019-08-12:
The rom I was using gave that exact md5: d41d8cd98f00b204e9800998ecf8427e, so I went and got the rom with the correct md5: b1774f03700dfc311bfd86411db629c3
Don't know why the previous rom gave that value, probably a bad dump, so I retested the new rom on the SNT and the bug is still happening in exactly the same way. 

This is what I'm doing: load the game, unplug controller in port 1, plug in hyperkin snes mouse (that always sends a button press input btw), then I do the steps to reproduce as described above and the sky glitches out. Maybe someone who owns both original and hyperkin snes mouse could test to see if somehow this could be triggered by the 3rd party mouse?

frederic-mahe, 2019-08-14:
Thanks for checking @Chronokeys 

There is an entry in SmokeMonster's pack for the SNES with the MD5 hash d41d8cd98f00b204e9800998ecf8427e and the name *Taekwon-Do* (Korea). This is probably a misnamed ROM then.

frederic-mahe, 2019-08-14:
As @Chronokeys suggested, this is what needs to be verified next:
> Maybe someone who owns both original and hyperkin snes mouse could test to see if somehow this could be triggered by the 3rd party mouse?

Rahzadan, 2019-08-14:
Snt Firmware:
Official 4.9

SD2SNES OG (non-Pro) - Firmware 1.10.3

I can confirm that this issue only seems to occur when using the Hyperkin SNES mouse, and it also happens on real hardware! This issue is not related to Super NT firmware in any way. I tested on my 2-1-3 SNES. See the video I made below:

[https://youtu.be/9CBeCxE1I98](url)

frederic-mahe, 2019-08-15:
Thanks @Rahzadan ! To summarize, the issue occurs on the Super Nt and the original SNES when using the Hyperkin SNES mouse, but not when using an original SNES mouse. This is not specific to the Jailbreak firmware or to the Super Nt, but rather an issue with the Hyperkin SNES mouse.

closed 2019-08-15 by frederic-mahe