SmokeMonsterPacks/Super-NT-Jailbreak #143: Width and Height Settings Lost (Issue has returned in 6.9

silentq, 2019-08-11:
Expected behavior and actual behavior
Width and Height Settings are not saving exactly like this issue from a while back.
https://github.com/SmokeMonsterPacks/Super-NT-Jailbreak/issues/1

Just as reported here when you change one setting (tick up the width from your saved setting after have rebooted the console) everything snaps back into place and your setting is restored.

Steps to reproduce the problem
1. Set your width and height
2. Save Setting
3. Turn Power Off
4. Turn Power On

(Setting you saved will display in in the width and height menu but not display in the manner you have it set to until you alter it.) 

- firmware version: 
JB6.9 2019-08-09

rubikcuber, 2019-08-11:
I have the same issue. I have a workaround that is working for me. If I set the system to boot to cart, rather than boot to menu, it does correctly apply my video settings. Similarly if I do "Run Cartridge" before selecting a ROM (including the in-built Turricans) it also applies my settings. I don't need a cartridge in for the "boot to cart" workaround to work.

dbnhorchner, 2019-08-11:
I've encountered a similar issues with scanlines (JB 6.9)

1. Enable "Normal Scanlines"
2. Save Settings
3. Turn Power off
4. Turn Power on
5. Scanlines are not active, but the GUI indicates that they are.

Nigoli, 2019-08-11:
I can also confirm. I use and prefer 720p because it has perfect integer scaling using square pixels.

As a temp solution, I usually go to the width and height settings and go down once to highlight Horizontal Position and then it’ll correct itself.

vdorza, 2019-09-24:
Yes... exactly the same here. Very strange

rtrind, 2019-10-03:
Same here. It loads in my resolution (720p), but the width/height/scanlines settings (which is all I have altered in the configuration) is only activated after I change anything on the configurations.

sbrade4, 2020-01-09:
I’m still hoping this issue is fixed outside of using the workaround. I figured something with the new fx games or dac introduced the issue. Does everyone have the same experience is what I’m wondering?  Like do some people not have this problem?

Beliaal, 2020-02-09:
Hi,

I had a similar issue, but i got it working as expected again simply by re-installing the 6.9 firmware and when completed i manually did a factory reset and rebooted. Worked like a charm afterwards, all settings was saved correctly, as expected. Probably unrelated, but somehow the 8bitdo BT connection issues was also greatly improved.

My exact steps:
1. I made a copy/backup of the SD-card to my computer
2. I formatted the SD-card using fat32format (http://www.ridgecrop.demon.co.uk/index.htm?guiformat.htm), 16k Allocation size, Quick format unchecked (Took about an hour).
3. I downloaded FW6.9, unpacked to the SD-card, and when completed, i ejected the SD from the computer.
4. Booted the SD-card and waited for the firmware to complete (2-4 minutes or so).
5. Finally i did a "Restore Factory Defaults" (Menu: "Save/Clear Settings") and rebooted.
6. Restore the original content back to the SD. PROFIT!

If you ask me, i don't think the steps 1-4 are that important, i am pretty confident a factory restore is what did the trick. Not sure what fixed the 8bitdo connection issues, but likely by moving the Super NT ever so slightly when i removed the SD was enough. We are talking about moving it about 1cm or so, but somehow it did the trick. Cant rule out factory settings perhaps, but i would not bet on it...

Hope this helps!

Ulisses-Medeiros, 2020-03-02:
the same happens to me, I tried with jailbreak version 6.9 and as the official 4.9, I installed version 4.9 and did the test the same happens if you start the super turrican special or 2, I tried what Beliaal proposed, and it didn't work. To play it is still necessary to change at least one of the settings to load my previously saved settings.

firmware version:
JB6.9 2019-08-09
OFW 4.9

eturquin, 2020-11-01:
Hi, anyone knows if this is or has been looked into, and if there is any chance that we will see another firmware update for the Super NT that may fix this issue? It's a pretty annoying regression tbh...

rubikcuber, 2020-11-03:
I'm still using the workaround which is to boot to a cartridge on startup. This "fixes" the issue for me, but it's a bit annoying (although I find it less annoying than having to go into the settings menu each time).

eTernal1911, 2020-11-23:
16 months and still no fix. :(

eturquin, 2021-02-23:
Looks like it has been fixed in the official 5.0 firmware, huzzah!!

closed 2021-02-23 by SmokeMonsterPacks