SmokeMonsterPacks/Super-NT-Jailbreak #144: Hori Controller (MUTE BUTTON) and SuperGameboy 2

FokinoPelvis, 2019-08-27:
I´m using 6.9  firmware with a SuperGameboy 2 and Hori Commander Gamepad
There is an issue about Cartridge audio using this gamepad.
Cartridge audio runs perfect but the Hori commander "MUTE" button (wich disable/enable sgb cartridge audio when u push it) don´t do anything. 
Maybe the enable audio cartridge checkbox, (supernt audio settings) make the sound always on, and force to don´t be disabled using the gamepad.

So, due to Super NT actual firmware limitations, it´s not possible to enable/disable the cartridge audio using the Hori controller. It´s possible to fix this in future revisions? Thank u!

FokinoPelvis, 2019-08-27:
Thank u for labeling Frederic!

SmokeMonsterPacks, 2021-02-24:
This didn't show up in the latest update so I'm closing the request to focus attention on bugs.

closed 2021-02-24 by SmokeMonsterPacks

bobbobbio, 2021-03-02:
I still have this problem actually on the non jail-broken firmware 5.0, so I'm surprised to see you saw if fixed on the jail-broken version