SmokeMonsterPacks/Super-NT-Jailbreak #147: SD card browser gets stuck when opening directory

mankeli, 2019-09-28:
SD card browser seems to get stuck when entering SPC directory with following files. (sorry about long list, I suspect there's some illegal filename there)
Game keeps running, cursor flashes but no input is read anymore.

3_Ninjas_Kick_Back.sp2
3x3_Eyes_Juuma_Houkan.sp2
3x3_Eyes_Seima_Kourinden.sp2
4_Nin_Shogi.sp2
7th_Saga.sp2
'96_Zenkoku_Koukou_Soccer_Senshuken.sp2
AAAHH!!!_Real_Monsters.sp2
ABC_Monday_Night_Football.sp2
A_Bug's_Life_(Super_Gameboy).sp2
Accelebrid.sp2
Ace_o_Nerae!.sp2
ACME_Animation_Factory.sp2
Acrobat_Mission.sp2
Action_Pachio.sp2
Actraiser_2.sp2
Actraiser.sp2
Addams_Family_Pugsley's_Scavenger_Hunt.sp2
Addams_Family_Values.sp2
Adventure_Island.sp2
Adventures_of_Dr._Franken.sp2
Aero_the_Acro-Bat_2.sp2
Aero_the_Acro-Bat.sp2
Aging_Cassette.sp2
Aguri_Suzuki_F-1_Super_Driving.sp2
Air_Cavalry.sp2
Air_Management_2_Koukuu_Ou_wo_Mezase_(Aerobiz_Supersonic).sp2
Air_Management_Ozora_ni_Kakeru_(Aerobiz).sp2
Akazukin_Cha_Cha.sp2
Aladdin.sp2
Albert_Odyssey_2.sp2
Albert_Odyssey.sp2
Alcahest.sp2
Alfred_Chicken.sp2
Alice's_Paint_Adventure.sp2
Alien_3.sp2
Alien_vs._Predator.sp2
Al_Unser_Jr's_Road_to_the_Top.sp2
American_Battle_Dome.sp2
American_Gladiators.sp2
An_American_Tail_Fievel_Goes_West.sp2
Andre_Agassi_Tennis.sp2
Animaniacs.sp2
Animaniacs_(Super_Gameboy).sp2
Aoki_Densetsu_Shoot!.sp2
Apocalypse_II.sp2
Appleseed_Prometheus_no_Shintaku.sp2
Arabian_Nights.sp2
Araiguma_Rascal.sp2
Archer_MacLean's_Dropzone.sp2
Arcus_Spirits.sp2
Ardy_Lightfoot.sp2
A_Ressha_de_Ikou_3_Super_Version.sp2
Aretha_2.sp2
Aretha.sp2
Arkanoid_Doh_it_Again.sp2
Armor_Police_Metal_Jack.sp2
Art_of_Fighting.sp2
Aryol.sp2
Asahi_Shinbun_Rensai.sp2
Asameshimae_Nyanko.sp2
Ashita_no_Joe.sp2
A.S.P._Air_Strike_Patrol.sp2
Asterix_&_Obelix.sp2
Asterix.sp2
Astral_Bout_2.sp2
Astral_Bout_3.sp2
Astrohawk_May_2003_Version.sp2
[... 1431 lines cut ...]
Yuujin_no_Furi_Furi_Girls.sp2
Yuuyu_no_Quiz_de_Go!_Go!.sp2
Yuu_Yuu_Hakusho_2.sp2
Yuu_Yuu_Hakusho.sp2
Yuu_Yuu_Hakusho_Tokubetsu-hen.sp2
Zakuro_no_Aji.sp2
Zan_2_Spirits.sp2
Zan_3_Spirits.sp2
Zenkoku_Juudan_Ultra_Shinri_Game.sp2
Zenkoku_Koukou_Soccer_Senshuken_2.sp2
Zenkoku_Koukou_Soccer_Senshuken.sp2
Zen_Nihon_GT_Sensyuken.sp2
Zennihon_Pro_Wrestling_2_-_3.4_Budoukan.sp2
Zennihon_Pro_Wrestling_Dash_-_Sekai_Saikyou_Tag_(Natsume_Championship_Wrestling).sp2
Zennihon_Pro_Wrestling.sp2
Zero_4_Champ_RR.sp2
Zero_the_Kamikaze_Squirrel.sp2
Z-Gundam_Away_to_the_Newtype.sp2
Zico_Soccer.sp2
Zig_Zag_Cat.sp2
Zombies_Ate_My_Neighbors.sp2
Zone.sp2
Zool.sp2
Zoop.sp2
Zootto_Mahjong!.sp2

rykerrk, 2019-09-28:
Maybe it's because there's 1523 files in one directory.

mankeli, 2019-09-29:
@rykerrk your comment doesn't really add any value, this is a bug report

rykerrk, 2019-09-29:
It's not a valid bug because you're not following the introductory notes on using the SD file browser.  I'll google this for you I guess:

*The maximum number of files (ROMs and subfolders) that can be placed in a given folder is around 300-500, depending on the length of the filenames.

Don't dump the entire Internet onto one directory and you'll probably find it works much better.

frederic-mahe, 2019-09-29:
Thanks @rykerrk . @mankeli please try to spread your files over several folders. Hopefully, it should fix the issue.

mankeli, 2019-09-29:
@rykerrk sure, but whether it is or is not a bug report in your eyes, you are still an arrogant little prick.

Anyway, sorry for noise! Missed that bit from the readme. (already spread my files of course..) I guess tho it could say when it's overflowing, but can't have everything.

frederic-mahe, 2019-09-30:
Let's keep it civil here. Thanks for everybody's input, I am closing the issue.

closed 2019-09-30 by frederic-mahe