SmokeMonsterPacks/Super-NT-Jailbreak #151: Super Double Dragon / Return of Double Dragon (graphic problem in elevator stage)

BarbieOnWeed, 2019-10-17:
## Expected behavior and actual behavior
In the elevator in the beginning of the game, a line of pixels are showing
to the upper right what should not be there.

https://youtu.be/b5zhu8hqaHo?t=128

## Steps to reproduce the problem
Start the game and play to the elevator stage
in the beginning of the game.

Both in PAL and NTSC mode.

## ROM name and hash value
Return of Double Dragon (Japan).sfc
CRC32: 10470B58

Super Double Dragon (Europe).sfc
CRC32: B3F6009B

Super Double Dragon (USA).sfc
CRC32: 09ED12D5

Return of Double Dragon (Japan) (original cart)

Super Double Dragon (Europe) (original cart)

- firmware version: JB6.9

frederic-mahe, 2019-10-17:
> a line of pixels are showing to the upper right what should not be there.

@BarbieOnWeed do you confirm that this line does not appear on a real Super Nintendo?

frederic-mahe, 2019-10-17:
Also, I have a slightly different CRC32 for _Super Double Dragon_ (USA):
```
bcced1be76ef920b562a555696bcb4583d1c8cea4d4b057cab6e0e09be8ef8c4
5dcaee58ba92e620055e1cee7f1c84da4e22a61a
38129cf359c7bc93a82358fa305b07ae
9ed12a5  # where you have 9ED12D5
```

BarbieOnWeed, 2019-10-17:
It does the same thing on a SuperCIC moded Super Famicom.
It was late and i only tested it on Super NT. Sorry... but now we know its not the NT. :)

frederic-mahe, 2019-10-17:
Thanks @BarbieOnWeed I'll try to test on my 1CHIP Super Famicom this weekend, just to double-check.

BarbieOnWeed, 2019-10-17:
> 
> 
> Also, I have a slightly different CRC32 for _Super Double Dragon_ (USA):
> 
> ```
> bcced1be76ef920b562a555696bcb4583d1c8cea4d4b057cab6e0e09be8ef8c4
> 5dcaee58ba92e620055e1cee7f1c84da4e22a61a
> 38129cf359c7bc93a82358fa305b07ae
> 9ed12a5  # where you have 9ED12D5
> ```

I made an error, mine to was 9ed12a5. Sorry it was late and I was tired.

frederic-mahe, 2019-10-19:
The glitch is also present on an original Super Famicom 1CHIP-2, so it is not an issue with the Super Nt.

closed 2019-10-19 by frederic-mahe