SmokeMonsterPacks/Super-NT-Jailbreak #153: Scanline issue

joegaff76, 2019-11-18:
I'm experiencing a problem with scanlines (JB 6.9)

Enable "Normal Scanlines" or "Hybrid Scanlines"
Save Settings
Turn Power off
Turn Power on
Scanlines are not active, but the GUI indicates that they are (along with all of my saved settings).  Only when I re-choose my scanline-type setting is when will they reappear.

romevi, 2019-11-19:
Are you playing games off the SD card or cartridge? I get this issue too, but only when playing off the SD card. If I turn off and turn back on the Advanced settings options, all my settings reapply, but they're always applied when running a game off the cartridge.
Annoying, but only takes about five seconds to adjust.

joegaff76, 2019-11-27:
Thanks for your reply.  It happens off the SD Card.  It's annoying and easy to adjust, but I thought I'd report it anyway as it is a bug.
Thanks again, Happy Thanksgiving!

romevi, 2019-12-06:
> Thanks for your reply.  It happens off the SD Card.  It's annoying and easy to adjust, but I thought I'd report it anyway as it is a bug. Thanks again, Happy Thanksgiving!

Actually, now that you brought it up, I noticed it's not just a bug when playing a game through the SD card. The Analogue logo in the background should have the settings applied, but it won't unless you do what I mentioned above. So I wonder if it's a bug with the original firmware and not the jailbroken one. I was running a ROM hack and went to do the usual fix, and I noticed the Analogue logo in the background didn't have the scanlines applied until I unselected/reselected Advanced Video Mode. Weird how I never noticed that before.

joegaff76, 2019-12-20:
Yes, it's an easy-to-deal-with but definitely an annoying bug.  Hopefully someone will discover the cause and correct it.
Thanks again!

Moonbase1, 2020-01-17:
Confirmed this happens with newest update.

fraggore, 2020-03-11:
I am having the same thing

MrGaussRA, 2020-10-22:
I can also confirm this is present in the latest JB firmware. This will only happen if you select a game from the SD card. It won't happen with the pre loaded Turrican ROM.

tjanas, 2021-02-23:
Can you retest and see if this was resolved with the latest firmware?

MrGaussRA, 2021-04-01:
After testing I can now confirm that the issue was solved in the latest update :D

closed 2021-04-01 by tjanas