SmokeMonsterPacks/Super-NT-Jailbreak #155: Send save RAM back to cart with CopySNES

BarbieOnWeed, 2020-01-11:
Any chance for an updated version of CopySNES to have the ability to transfer saves back to the cart.
Have alot of old emulator saves and on Super NT I would love to use on original carts. :)

GunbusterZero, 2020-01-12:
If you can't wait for this feature like me, I would recommend looking into the Retrode 2, with this device it is possible to copy almost all saves back to an snes cart.

JediWed, 2020-01-20:
[x] Vote for this :-)

closed 2020-03-07 by SmokeMonsterPacks

Drakonas, 2020-03-07:
> If you can't wait for this feature like me, I would recommend looking into the Retrode 2, with this device it is possible to copy almost all saves back to an snes cart.

Before you willy nilly recommend a product I would do research first. The Retrode 2 doesn't support most special chip games, specifically any where the special chip manipulates SRAM (which is, again, most). If you want this feature 100% you'll instead want a fully-outfitted Sanni Cart reader, but they are diy projects only. You'll have to find someone that can build it for you if you can't.

tjanas, 2023-07-11:
@BarbieOnWeed @SaberSage @turcane @Drakonas please retest with [JB v7.2](https://github.com/analoguejb/Analogue-Super-Nt-JB). This feature has been added.