SmokeMonsterPacks/Super-NT-Jailbreak #157: Launching "Play SPCs" from the Tools menu may disrupt color-changing LED pattern on Super Nt

tjanas, 2020-03-01:
Using firmware JB6.9. Initial condition: LED settings on Super Nt are configured other than Standby Color, where the LED on the console is not a constant color (colors animate at some non-zero speed). When launching the SPC Player core via the "Play SPCs" menu option, the LED on the Super Nt will not animate properly anymore (it will have periods where the LED is not lit at all). Going into settings and tweaking LED settings will restore normal behavior of the LED. This has been reported and confirmed by two users in the classic gaming discord.

re4mat, 2021-02-24:
Appears to still be present in JB7.0.

tjanas, 2021-02-25:
I also confirmed this issue still exists with latest firmware. 

https://user-images.githubusercontent.com/20304619/109103261-2cc0e600-76f0-11eb-96a4-acd85e71429f.MOV

tjanas, 2023-07-11:
Reported as fixed with [JB v7.2](https://github.com/analoguejb/Analogue-Super-Nt-JB).

tjanas, 2023-07-12:
I have confirmed this to be fixed.

closed 2023-07-12 by tjanas