SmokeMonsterPacks/Super-NT-Jailbreak #159: FFV garbled cursor and character graphics

mattysaurus, 2020-03-08:
Firmware 4.9, No-intro Japanese rom and GBA script romhack

Expected behavior:
When entering a shop (Tule, for example) and talking to the shopkeeper, both the character and character sprites appear as expected)

Actual behavior: Occasionally, randomly, the cursor hand graphics are garbled and the characters when browsing weapons and armor are garbled as well. 

## Steps to reproduce the problem
Enter a shop and speak to a shopkeeper. You may have to enter or exit several times. It should appear at least once every dozen times. Sometimes much more often. Interestingly, I could not replicate the issue when the Region was selected as PAL (of course the game plays more slowly this way). 

## ROM name and hash value
Final Fantasy V (Japan).sfc
No-Intro

## firmware

- firmware version: 4.9, no jailbreak applied, using Super Everdrive 2.1

## relevant Super Nt settings
Does not occur when set to PAL

mattysaurus, 2020-03-08:
[image]

Here is a photo of glitch.

redacted173, 2020-08-07:
I haven't been able to reproduce this on snt, but it looks very similar to the ff6 dma bug.  If I skip the DMA at $C2DA12 during the shop transition in an emulator I get the exact same tile pattern.  The DMA writes new sprite data at VRAM (byte address) $8000 for the shop.

[image]

redacted173, 2020-08-08:
I was able to hit the problem on snt with the attached zip with floating bps patch to ff5 jp on a SD2SNES with all the options disabled in the SD2SNES menu.  It only happened on the snt 4.9 (non-JB) firmware for me.  I couldn't hit it on 6.9, but I think it still has the bug.  Just go in and out of the shop menu a few times until the sprites are messed up.

CRC-32: C1BC267D
SHA-256: C6858D5C02894A6CC71F4DD452C7F288B319D1952CA56FDB185B4BF5E26244A2

Looks like the same problem as ff6: https://github.com/SmokeMonsterPacks/Super-NT-Jailbreak/issues/27.  DMA sending over the sprites to VRAM ends early and all the stale tile data from before the shop menu is used.  Trigger on the $420B write at $C2DA12 or the spurious(?) read at what I think is $0E8003 or $CE8003.  I ran out of probes to get the upper bank bits.  Can hopefully fix that and get some more address bit and other control signals soon.

[image]

[ff5jp_snt_issue159.zip](https://github.com/SmokeMonsterPacks/Super-NT-Jailbreak/files/5044523/ff5jp_snt_issue159.zip)

re4mat, 2021-02-24:
Tested on JB7.0 using JP ROM patched with redacted173's patch above. Did not hit the bug. (Have not tried with the vanilla ROM, though.)

tjanas, 2021-05-08:
@mattysaurus can you reproduce on latest JB version?

tjanas, 2023-05-05:
@redacted173 can you see if it is reproducible on the latest official firmware update from today?

redacted173, 2023-05-09:
I can't reproduce the bug on SNT firmware 5.1.

tjanas, 2023-05-09:
Closing this issue. @mattysaurus let us know if you experience this issue again with https://www.analogue.co/support/super-nt/firmware/5.1

closed 2023-05-09 by tjanas