SmokeMonsterPacks/Super-NT-Jailbreak #160: Feature Request: SA1 and SuperFx 1 & 2 support

ChloeMarieTaylor76, 2020-04-12:
Hi, apologies for placing it here, I didn't know of anywhere else to ask. The last firmware added support for some special chips, I know playing Mario Kart again brought much joy. Is there an plans to add support for SA1 and SuperFx chips as some of the best games used this. I check back at least once as week hoping there might be support added. Analogue's attention seems to have been drawn away from the SuperNT and MegaSg both of which I own. Is there any plan to add support please? It seems mister and certain carts have fpga support for these chips already. I guess I have the same question for the SVP on the MegaSg.

SamHard, 2020-05-23:
Unfortunately, the "word on the street" is that this is not possible on the Super NT hardware.
"Kevtris has already said (in a live stream iirc) that the FPGA hardware in the Super NT isn't powerful enough to do Super FX or SA-1" ([source](https://www.reddit.com/r/snes/comments/cobmpr/super_nt_special_chip_support_new_jailbreaks_are/ewomtnt?utm_source=share&utm_medium=web2x))

ChloeMarieTaylor76, 2020-05-23:
I thought the Super NT had a cyclone V to the sd2snes' Cyclone IV. I guess it means there is not enough space left to fit in the logic for the SA1 or super FX, rather than it is not powerful enough :( The mister snes core supports all the chips and uses a cyclone V on a Terasic DE10-Nano. Well I live in hope :( Shame. I'd like everything to be built into the super nt rather than having to switch in carts all the time.

Drakonas, 2020-05-24:
> I thought the Super NT had a cyclone V to the sd2snes' Cyclone IV.

There are many variants of a specific Cyclone model. SD2SNES does not include all the SNES in its programming... That takes a lot of space on an FPGA. The Super Nt has a Cyclone V 5CEA4 variant which has 49,000LE (18480ALM). The Cyclone 5 variant for the MiSTer had 110,000LE (41,500ALM). As you can see, the one on the MiSTer is more than half the size. The Super Nt has ran out of room unfortunately for large-scale things like the SA-1 which is similar to the same CPU as the Super Nintendo main CPU, running in parallel (they're able to interrupt each other), with a few extra features.

ChloeMarieTaylor76, 2020-05-24:
Thank you for explaining, I thought it must be an issue of space. :( So upsetting really. I wanted to play a lot of the SA1 modded roms to fix some of the terrible slow down issues with the likes of super R-type. There might be a market opportunity for an fpga special chip cartridge upgrade for the super nt. Catch 22 now, there is the mister project but you lose the analogues sleek case, or stuck with the super NT which now seems incomplete. Bummer

Drakonas, 2020-05-25:
> ... the super NT which now seems incomplete. Bummer
I question how your roms accessible to you make the Super Nt "incomplete" when the Super Nt's official firmware is only intended to run cartridges. The hacked firmware should be considered a blessing that it ever even happened in my opinion. If you want certain games, you can still buy them if you buy the cartridges... It's not like the Super Nt can't play SA-1 games.

ChloeMarieTaylor76, 2020-05-25:
It does seem defensive. yes. In my use-case it is incomplete. I wanted the convenience of not having to switch cartridges all the time, using my games from the SD card rather than add wear and tear to cartridges approaching nearly 30 years old. I wanted to play fixed roms of games that had issues such as the chronic slowdown that the snes often had. It is hard to do that when quite a few of the classics are not able to counted in that use-case. The fact is there is a custom firmware and it was the reason I bought it (worst kept secret on the planet), most probably like lots of other people, and I doubt it did Analogue any harm in sales whatsoever, quite the contrary. It is somewhat hard to believe the firmware could be customised to this extend without Analogue's blessing.

TheRealFlamepanther, 2020-05-25:
> There might be a market opportunity for an fpga special chip cartridge upgrade for the super nt.

You just described an FXPAK ;)

Drakonas, 2020-05-25:
> It does seem defensive. yes. In my use-case it is incomplete. I wanted the convenience of not having to switch cartridges all the time, using my games from the SD card rather than add wear and tear to cartridges approaching nearly 30 years old. I wanted to play fixed roms of games that had issues such as the chronic slowdown that the snes often had. It is hard to do that when quite a few of the classics are not able to counted in that use-case. The fact is there is a custom firmware and it was the reason I bought it (worst kept secret on the planet as it is mentioned in nearly every review). Most probably like lots of other people, and I doubt it did Analogue any harm in sales whatsoever, quite the contrary. It is somewhat hard to believe the firmware could be customised to this extent without Analogue's blessing, they did nothing in later firmware upgrades to prevent it. I am not having a dig at anyone and I am grateful for the custom firmwares, it is just a real shame there is this hardware limitation.

Why not buy a SD2SNES then? It does all that and supports nearly every game.

ChloeMarieTaylor76, 2020-05-25:
> 
> 
> > There might be a market opportunity for an fpga special chip cartridge upgrade for the super nt.
> 
> You just described an FXPAK ;)

Looks like the neatest way to go for me. Quick question please if anyone has experience of this, are you able to put in the FXPAK and load the games from the superNT's menu or would it be a case of booting automatically to the FXPAK's and loading all games from there?

ChloeMarieTaylor76, 2020-05-25:
Have ordered a Mister setup, guess my super nt and mega sg will be going on ebay. They seem a bit redundant now, nice as they are. My hope is the analogue pocket will effectively be a Mister with a professional standard case :)

frederic-mahe, 2020-05-25:
Thanks everyone for taking the time to open and discuss this issue, I am going to close it for now.

closed 2020-05-25 by frederic-mahe

Drakonas, 2020-05-26:
> > > There might be a market opportunity for an fpga special chip cartridge upgrade for the super nt.
> > 
> > 
> > You just described an FXPAK ;)
> 
> Looks like the neatest way to go for me. Quick question please if anyone has experience of this, are you able to put in the FXPAK and load the games from the superNT's menu or would it be a case of booting automatically to the FXPAK's and loading all games from there?
> 
> The thought occurs for the £200 I could just sell both my analogue mega sg and analogue super nt, and buy a Mister setup and be £300 + the £200 up!

FXPAK (previously named SD2SNES) had its own menu and in my opinion it's better than the Super Nt's. The menu is themeable too with actual bmp graphics. It also has a "recent roms" list and supports cheats. So yes, you can just autoboot to it and be set. It also supports games with RTC's and almost all special chip games.